tork-a/rtmros_nextage #295: Running Ueye error

shzargar, 2017-02-15:
roslaunch nextage_ros_bridge nextage_ueye_stereo.launch 

gives me this error. It seems that I don't have this library /opt/ros/indigo/lib//libueye_cam_nodelet.so . I don't know how to add it.

... logging to /home/shzargar/.ros/log/6e50918a-f3d0-11e6-a3d9-98e7f45b297c/roslaunch-Nxtag-5141.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Nxtag:45590/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /ueye_cam_nodelet_master/auto_exposure: False
 * /ueye_cam_nodelet_master/auto_frame_rate: False
 * /ueye_cam_nodelet_master/auto_gain: True
 * /ueye_cam_nodelet_master/auto_white_balance: True
 * /ueye_cam_nodelet_master/binning: 2
 * /ueye_cam_nodelet_master/blue_gain: 16
 * /ueye_cam_nodelet_master/camera_id: 1
 * /ueye_cam_nodelet_master/camera_intrinsics_file: 
 * /ueye_cam_nodelet_master/camera_name: right
 * /ueye_cam_nodelet_master/camera_parameters_file: 
 * /ueye_cam_nodelet_master/camera_topic: image_raw
 * /ueye_cam_nodelet_master/color_mode: rgb8
 * /ueye_cam_nodelet_master/exposure: 33
 * /ueye_cam_nodelet_master/ext_trigger_mode: False
 * /ueye_cam_nodelet_master/flash_delay: 30000
 * /ueye_cam_nodelet_master/flash_duration: 1000
 * /ueye_cam_nodelet_master/frame_rate: 30.0
 * /ueye_cam_nodelet_master/gain_boost: False
 * /ueye_cam_nodelet_master/green_gain: 1
 * /ueye_cam_nodelet_master/image_height: 1200
 * /ueye_cam_nodelet_master/image_left: -1
 * /ueye_cam_nodelet_master/image_top: -1
 * /ueye_cam_nodelet_master/image_width: 1600
 * /ueye_cam_nodelet_master/master_gain: 0
 * /ueye_cam_nodelet_master/pixel_clock: 24
 * /ueye_cam_nodelet_master/red_gain: 0
 * /ueye_cam_nodelet_master/sensor_scaling: 1.0
 * /ueye_cam_nodelet_master/subsampling: 1
 * /ueye_cam_nodelet_master/white_balance_blue_offset: 0
 * /ueye_cam_nodelet_master/white_balance_red_offset: 0
 * /ueye_cam_nodelet_slave1/auto_exposure: False
 * /ueye_cam_nodelet_slave1/auto_frame_rate: False
 * /ueye_cam_nodelet_slave1/auto_gain: True
 * /ueye_cam_nodelet_slave1/auto_white_balance: True
 * /ueye_cam_nodelet_slave1/binning: 2
 * /ueye_cam_nodelet_slave1/blue_gain: 16
 * /ueye_cam_nodelet_slave1/camera_id: 2
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file: 
 * /ueye_cam_nodelet_slave1/camera_name: left
 * /ueye_cam_nodelet_slave1/camera_parameters_file: 
 * /ueye_cam_nodelet_slave1/camera_topic: image_raw
 * /ueye_cam_nodelet_slave1/color_mode: rgb8
 * /ueye_cam_nodelet_slave1/exposure: 33
 * /ueye_cam_nodelet_slave1/ext_trigger_mode: False
 * /ueye_cam_nodelet_slave1/flash_delay: 0
 * /ueye_cam_nodelet_slave1/flash_duration: 1000
 * /ueye_cam_nodelet_slave1/frame_rate: 30.0
 * /ueye_cam_nodelet_slave1/gain_boost: False
 * /ueye_cam_nodelet_slave1/green_gain: 1
 * /ueye_cam_nodelet_slave1/image_height: 1200
 * /ueye_cam_nodelet_slave1/image_left: -1
 * /ueye_cam_nodelet_slave1/image_top: -1
 * /ueye_cam_nodelet_slave1/image_width: 1600
 * /ueye_cam_nodelet_slave1/master_gain: 0
 * /ueye_cam_nodelet_slave1/pixel_clock: 24
 * /ueye_cam_nodelet_slave1/red_gain: 0
 * /ueye_cam_nodelet_slave1/sensor_scaling: 1.0
 * /ueye_cam_nodelet_slave1/subsampling: 1
 * /ueye_cam_nodelet_slave1/white_balance_blue_offset: 0
 * /ueye_cam_nodelet_slave1/white_balance_red_offset: 0

NODES
  /left/
    image_proc (image_proc/image_proc)
  /right/
    image_proc (image_proc/image_proc)
  /
    head_link_to_stereo_link (tf/static_transform_publisher)
    left_force_subscribe (rostopic/rostopic)
    nodelet_manager (nodelet/nodelet)
    right_force_subscribe (rostopic/rostopic)
    stereo_image_proc (stereo_image_proc/stereo_image_proc)
    stereo_link_to_stereo_frame (tf/static_transform_publisher)
    ueye_cam_nodelet_master (nodelet/nodelet)
    ueye_cam_nodelet_slave1 (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [5153]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6e50918a-f3d0-11e6-a3d9-98e7f45b297c
process[rosout-1]: started with pid [5166]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [5182]
process[ueye_cam_nodelet_master-3]: started with pid [5184]
process[ueye_cam_nodelet_slave1-4]: started with pid [5185]
process[left_force_subscribe-5]: started with pid [5186]
process[right_force_subscribe-6]: started with pid [5187]
process[right/image_proc-7]: started with pid [5194]
process[left/image_proc-8]: started with pid [5213]
process[stereo_image_proc-9]: started with pid [5214]
process[head_link_to_stereo_link-10]: started with pid [5217]
process[stereo_link_to_stereo_frame-11]: started with pid [5218]
[ INFO] [1487198721.762002716]: Initializing nodelet with 4 worker threads.
WARNING: topic [/left/camera_info] does not appear to be published yet
WARNING: topic [/right/camera_info] does not appear to be published yet
[ERROR] [1487198721.925323027]: Failed to load nodelet [/ueye_cam_nodelet_slave1] of type [ueye_cam/ueye_cam_nodelet] even after refreshing the cache: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
[ERROR] [1487198721.925469816]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
[FATAL] [1487198721.925669610]: Failed to load nodelet '/ueye_cam_nodelet_slave1` of type `ueye_cam/ueye_cam_nodelet` to manager `nodelet_manager'
[ERROR] [1487198722.064550624]: Failed to load nodelet [/ueye_cam_nodelet_master] of type [ueye_cam/ueye_cam_nodelet] even after refreshing the cache: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
[ERROR] [1487198722.064591290]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
[FATAL] [1487198722.064691268]: Failed to load nodelet '/ueye_cam_nodelet_master` of type `ueye_cam/ueye_cam_nodelet` to manager `nodelet_manager'
[ueye_cam_nodelet_slave1-4] process has died [pid 5185, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_slave1 __log:=/home/shzargar/.ros/log/6e50918a-f3d0-11e6-a3d9-98e7f45b297c/ueye_cam_nodelet_slave1-4.log].
log file: /home/shzargar/.ros/log/6e50918a-f3d0-11e6-a3d9-98e7f45b297c/ueye_cam_nodelet_slave1-4*.log
[ueye_cam_nodelet_master-3] process has died [pid 5184, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_master __log:=/home/shzargar/.ros/log/6e50918a-f3d0-11e6-a3d9-98e7f45b297c/ueye_cam_nodelet_master-3.log].
log file: /home/shzargar/.ros/log/6e50918a-f3d0-11e6-a3d9-98e7f45b297c/ueye_cam_nodelet_master-3*.log

130s, 2017-02-15:
Could you make sure that you're running the launch above on the Ubuntu computer that the cameras' cables are connected to?

Necessary dependency should be installed automatically (defined [here](https://github.com/tork-a/rtmros_nextage/blob/c454a17c8385ddf3c7314e9ddb8f46763e50b4dd/nextage_ros_bridge/package.xml#L26)) as long as the tutorial was followed as it is. You can manually install the missing library by `apt-get install ros-indigo-ueye-cam` as a quick workaround, if it's still missing.

shzargar, 2017-02-21:
I've checked on the nextage computer, it seems one of the hand camera is not working. I will come back with the message.

My questions: is there any specification for cameras (resolution and ...)?
Is there any way to communicate with cameras via ethernet, I mean not using the nextage computer.

130s, 2017-02-22:
I'm not sure if I fully understand what you're asking but I think of 2 answers:

**Using cameras you have, instead of the default Ueye cameras**
I added a [subsection](http://wiki.ros.org/rtmros_nextage/Tutorials/Setup%20stereo%20camera#Using_your_own_cameras_on_HiroNXO) in the wiki page.

**Using the default Ueye cameras**

> is there any specification for cameras (resolution and ...)?

The aforementioned driver package [ueye_cam](http://wiki.ros.org/ueye_cam) should take care of connections. For this I also added [a few lines in the wiki](http://wiki.ros.org/rtmros_nextage/Tutorials/Setup%20stereo%20camera#Introduction).

> Is there any way to communicate with cameras via ethernet, I mean not using the nextage computer.

To avoid confusion let's call `UI Controller Box` (Nextage's Ubuntu) as Comp-A and your computer (assuming another Ubuntu) as Comp-B.

I assume it's possible to disconnect the cameras cable from Comp-A and connect to Comp-B. I don't know how that is easy/hard. Either way I suggest to make sure you take enough note about the current setting so that you can plug them back in when you want to.

Or if you want to keep the current setting and just want to access the cameras' output from Comp-B, this is where the distributed network architecture of ROS comes very handy -- run the camera "node"s on Comp-A, then from Comp-B you can subscribe to the camera "topic"s. A bit more concretely:

1. Comp-A; run `roslaunch nextage_ros_bridge nextage_ueye_stereo.launch`
2. Comp-B; set your `rosmaster` toward Comp-A by e.g.:

  ```
export ROS_MASTER_URI=http://%Comp-A's ip address%:11311
```

3. Comp-B; Subscribe to the camera topics (e.g. `rostopic list`, or visualize on RViz etc. This depends on what you want to do.)

Step 1 above is NEXTAGE-specific but the rest is standard ROS way to subscribe the topics published on a remote host.

Lastly, FYI related to cameras, calibration using head-camera is being worked out. https://github.com/tork-a/rtmros_nextage/issues/291

534o, 2017-02-23:
```
Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
```
you have to get libueye_api.so by manually install the ueye sdk from https://github.com/anqixu/ueye_cam/blob/master/cmake_modules/DownloadUEyeDriversUnofficial.cmake#L34

shzargar, 2017-02-24:
I tried to launch from Nextage's Ubuntu and I got this error. 

... logging to /home/nxouser/.ros/log/088dccfc-f3d8-11e6-a72d-000bab77d7ba/roslaunch-tork-desktop-27088.log
Checking log directory for disk usage. This may take awhile.Nextage's Ubuntu
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tork-desktop:46928/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion
 * /ueye_cam_nodelet_master/auto_exposure
 * /ueye_cam_nodelet_master/auto_frame_rate
 * /ueye_cam_nodelet_master/auto_gain
 * /ueye_cam_nodelet_master/auto_white_balance
 * /ueye_cam_nodelet_master/binning
 * /ueye_cam_nodelet_master/blue_gain
 * /ueye_cam_nodelet_master/camera_id
 * /ueye_cam_nodelet_master/camera_intrinsics_file
 * /ueye_cam_nodelet_master/camera_name
 * /ueye_cam_nodelet_master/camera_parameters_file
 * /ueye_cam_nodelet_master/camera_topic
 * /ueye_cam_nodelet_master/color_mode
 * /ueye_cam_nodelet_master/exposureNextage's Ubuntu
 * /ueye_cam_nodelet_master/ext_trigger_mode
 * /ueye_cam_nodelet_master/flash_delay
 * /ueye_cam_nodelet_master/flash_duration
 * /ueye_cam_nodelet_master/frame_rate
 * /ueye_cam_nodelet_master/gain_boost
 * /ueye_cam_nodelet_master/green_gain
 * /ueye_cam_nodelet_master/image_height
 * /ueye_cam_nodelet_master/image_left
 * /ueye_cam_nodelet_master/image_top
 * /ueye_cam_nodelet_master/image_width
 * /ueye_cam_nodelet_master/master_gain
 * /ueye_cam_nodelet_master/pixel_clock
 * /ueye_cam_nodelet_master/red_gain
 * /ueye_cam_nodelet_master/sensor_scaling
 * /ueye_cam_nodelet_master/subsampling
 * /ueye_cam_nodelet_master/white_balance_blue_offset
 * /ueye_cam_nodelet_master/white_balance_red_offset
 * /ueye_cam_nodelet_slave1/auto_exposure
 * /ueye_cam_nodelet_slave1/auto_frame_rate
 * /ueye_cam_nodelet_slave1/auto_gain
 * /ueye_cam_nodelet_slave1/auto_white_balanceNextage's Ubuntu
 * /ueye_cam_nodelet_slave1/binning
 * /ueye_cam_nodelet_slave1/blue_gain
 * /ueye_cam_nodelet_slave1/camera_id
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file
 * /ueye_cam_nodelet_slave1/camera_name
 * /ueye_cam_nodelet_slave1/camera_parameters_file
 * /ueye_cam_nodelet_slave1/camera_topic
 * /ueye_cam_nodelet_slave1/color_mode
 * /ueye_cam_nodelet_slave1/exposure
 * /ueye_cam_nodelet_slave1/ext_trigger_mode
 * /ueye_cam_nodelet_slave1/flash_delay
 * /ueye_cam_nodelet_slave1/flash_duration
 * /ueye_cam_nodelet_slave1/frame_rate
 * /ueye_cam_nodelet_slave1/gain_boost
 * /ueye_cam_nodelet_slave1/green_gainNextage's Ubuntu
 * /ueye_cam_nodelet_slave1/image_height
 * /ueye_cam_nodelet_slave1/image_left
 * /ueye_cam_nodelet_slave1/image_top
 * /ueye_cam_nodelet_slave1/image_width
 * /ueye_cam_nodelet_slave1/master_gain
 * /ueye_cam_nodelet_slave1/pixel_clock
[... 56 lines cut ...]
Blue Gain:		16
Gain Boost:		0
Auto Exposure:		0
Exposure (ms):		33.0081
Auto White Balance:	1
WB Red Offset:		0
WB Blue Offset:		0
Flash Delay (us):	30000
Flash Duration (us):	1000
Ext Trigger Mode:	0
Auto Frame Rate:	0
Frame Rate (Hz):	29.6665
Pixel Clock (MHz):	24
Mirror Image Upside Down:	0
Mirror Image Left Right:	0

[ERROR] [1487201988.125173399]: Could not start free-run live video moNextage's Ubuntude for [right] (IS_INVALID_BUFFER_SIZE)
[ERROR] [1487201988.125205025]: Shutting down driver nodelet for [right]
[FATAL] [1487201988.318650451]: Service call failed!
[ueye_cam_nodelet_slave1-4] process has died [pid 27167, exit code 255, cmd /opt/ros/hydro/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_slave1 __log:=/home/nxouser/.ros/log/088dccfc-f3d8-11e6-a72d-000bab77d7ba/ueye_cam_nodelet_slave1-4.log].
log file: /home/nxouser/.ros/log/088dccfc-f3d8-11e6-a72d-000bab77d7ba/ueye_cam_nodelet_slave1-4*.log
[nodelet_manager-2] process has died [pid 27127, exit code -11, cmd /opt/ros/hydro/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/nxouser/.ros/log/088dccfc-f3d8-11e6-a72d-000bab77d7ba/nodelet_manager-2.log].
log file: /home/nxouser/.ros/log/088dccfc-f3d8-11e6-a72d-000bab77d7ba/nodelet_manager-2*.log
[ueye_cam_nodelet_master-3] process has finished cleanly
log file: /home/nxouser/.ros/log/088dccfc-f3d8-11e6-a72d-000bab77d7ba/ueye_cam_nodelet_master-3*.log

shzargar, 2017-02-24:
As well my question about camera specification is about to know what type of camera is used for our nextage robot and what is its specification (like is it a color camera, resolution and frequency ...).

130s, 2017-02-25:
> [ERROR] [1487201988.125173399]: Could not start free-run live video mode for [right] (IS_INVALID_BUFFER_SIZE)

This error is likely solved with a version of camera driver software that was released awhile ago ([the issue ticket](https://github.com/anqixu/ueye_cam/issues/24) and [release was done afterward](https://github.com/anqixu/ueye_cam/issues/40)).

You might want to tell us the version of camera driver software on the computer you used. As of today the latest available is 1.0.15. For instance I get the following.

```
dpkg -p ros-indigo-ueye-cam | grep Ver
Version: 1.0.15-0trusty-20170130-044337-0800
$ rosversion ueye_cam
1.0.15
```

130s, 2017-02-25:
> As well my question about camera specification is about to know what type of camera is used for our nextage robot and what is its specification (like is it a color camera, resolution and frequency ...).

For the cameras bundled on the robot when it was shipped, the best resource I was able to find online is http://nextage.kawada.jp/en/stereovision/. Honestly I don't have much more info than that website provides, other than that the camera is called `Ueye` and made by IDS. Its software driver package is http://wiki.ros.org/ueye_cam.

@emijah @youtalk may be able to address your concern better.

130s, 2017-03-07:
Closing due to inactivity. If there's anything to add please reopen or just add comments.

closed 2017-03-07 by 130s