tork-a/rtmros_nextage #302: Impedance control seems not working for NEXTAGE Open on hrpsys-simulator

130s, 2017-03-27:
With the same set of commands from https://github.com/start-jsk/rtmros_hironx/pull/482, NEXTAGE in the hrpsys-simulator doesn't do anything specific to impedance control.

I found when I run from Python i/f:
```
In [1]: robot.startImpedance('larm')
```

the following occurs in `nextage_ros_bridge_simulation.launch` console:
```
[rtmlaunch] check connection/activation
[ic] Could not found impedance controller param [larm]
```

Same error for `rarm` with rarm param not found.

```
$ rtls localhost:15005/
StateHolderServiceROSBridge.rtc        fk.rtc                             DataLoggerServiceROSBridge.rtc  SequencePlayerServiceROSBridge.rtc
ImageSensorROSBridge_HandLeft.rtc      sh.rtc                             HiroNX(Robot)0.rtc              ImageSensorROSBridge_HeadRight.rtc
seq.rtc                                HrpsysJointTrajectoryBridge0.rtc   log.rtc                         HGcontroller0.rtc
sc.rtc                                 ModelLoader                        el.rtc                          CollisionDetector0.rtc
ImageSensorROSBridge_HandRight.rtc     ImageSensorROSBridge_HeadLeft.rtc  HrpsysSeqStateROSBridge0.rtc    longfloor(Robot)0.rtc
ForwardKinematicsServiceROSBridge.rtc  tork-kudu1.host_cxt/                ic.rtc                          rmfo.rt
```

k-okada, 2017-03-28:
> With the same set of commands from start-jsk/rtmros_hironx#482

please confirm if you have conf file with end_effector key like
https://github.com/start-jsk/rtmros_hironx/pull/482/files

130s, 2017-03-29:
Yes,
```
$ rospack find nextage_ros_bridge 
/home/rosnoodle/cws_hironxo/src/tork-a/rtmros_nextage/nextage_ros_bridge
$ cat `rospack find nextage_ros_bridge`/conf/nextage.conf                                    
model: file:///home/rosnoodle/cws_hironxo/src/tork-a/rtmros_nextage/nextage_ros_bridge/models/nextage.dae
dt: 0.005
collision_pair: RARM_JOINT2:LARM_JOINT2 RARM_JOINT3:LARM_JOINT3 RARM_JOINT4:LARM_JOINT4 RARM_JOINT5:LARM_JOINT5
end_effectors: rarm,RARM_JOINT5,CHEST_JOINT0,0.0,0.0,0.0,-0.57735027,-0.57735027,-0.57735027,-2.0943951023931953,larm,LARM_JOINT5,CHEST_JOINT0,0.0,0.0,0.0,0,1,0,-1.5707963267948966,
```

k-okada, 2017-03-29:
Does your robot have force torque sensors?
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceController.cpp#L199-L251