tork-a/rtmros_nextage #306: goInitial not waiting previous commands' execution

130s, 2017-03-29:
[Video](https://youtu.be/60ASIEsRsm8?t=1m45s)

[goInitial](http://docs.ros.org/indigo/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#a295b0b4950cb580273b224dc659c8a23) takes `wait`, which is [passed to setJointAnglesOfGroup](https://github.com/start-jsk/rtmros_hironx/blob/b48585cc03a942dbc0fd4aadb1996672b93d88b8/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L429) as well as calling [seq_svc.waitInterpolationOfGroup](https://github.com/start-jsk/rtmros_hironx/blob/b48585cc03a942dbc0fd4aadb1996672b93d88b8/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L434). But looking at the left arm in the video `goInitial` seems to be taking precedence over the previous commands.

```
term-1$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch
term-3$  ipython -i `rospack find nextage_ros_bridge`/script/nextage.py
:
(on ipython terminal)

import math

robot.goInitial(tm=3)
RARM_POSE_1_RAD=[math.radians(deg) for deg in [6.138721628714777, -3.5487984309652205, -120.55334918111939, 18.14685807022124, 34.26540672437406, 4.139800681231482]]
RARM_POSE_2_RAD=[math.radians(deg) for deg in [15.336949634902222, -17.93199578325557, -132.19955608003514, 30.563307862688106, 60.55984865690648, 11.400978128655172]]
RARM_POSE_3_RAD=[math.radians(deg) for deg in [28.708427899147136, -41.49811023362566, -126.83963439177212, 71.75247032171964, 76.12473596686232, 41.72122191283136]]

LARM_POSE_1_RAD=[math.radians(deg) for deg in [ -6.138721903931133, -3.5487808134794783, -120.55332035549783, -18.146854340540436, 34.265400698239354, -4.139771787955316]]
LARM_POSE_2_RAD=[math.radians(deg) for deg in [-15.344437570393968, -17.938658725579963, -132.20654620491194, -30.578241697857496, 60.57372124239448, -11.410352523705216]]
LARM_POSE_3_RAD=[math.radians(deg) for deg in [-28.689573310750337, -41.46971571358063, -126.8535520134371, -71.67740452950109, 76.11674434234673, -41.66291771527085]]

robot.playPatternOfGroup('rarm', [RARM_POSE_1_RAD, RARM_POSE_2_RAD, RARM_POSE_3_RAD], [1.5, 2, 3])
robot.playPatternOfGroup('larm', [LARM_POSE_1_RAD, LARM_POSE_2_RAD, LARM_POSE_3_RAD], [1.5, 2, 3])
robot.playPatternOfGroup('rarm', [RARM_POSE_2_RAD, RARM_POSE_1_RAD], [2, 1.5])
robot.goInitial()
```
```
$ rospack find nextage_ros_bridge
/opt/ros/indigo/share/nextage_ros_bridge
$ rospack find hrpsys
/opt/ros/indigo/share/hrpsys
$ rosversion hrpsys                                                   
315.10.1
$ rosversion hironx_ros_bridge
1.1.22
$ rosversion nextage_ros_bridge                                       
0.7.15
```

k-okada, 2017-04-07:
wait is not wait for previous commands ends, it will wait for current
command ends, so hope
```
robot.playPatternOfGroup('rarm', [RARM_POSE_1_RAD, RARM_POSE_2_RAD,
RARM_POSE_3_RAD], [1.5, 2, 3])
robot.playPatternOfGroup('larm', [LARM_POSE_1_RAD, LARM_POSE_2_RAD,
LARM_POSE_3_RAD], [1.5, 2, 3])
robot.playPatternOfGroup('rarm', [RARM_POSE_2_RAD, RARM_POSE_1_RAD], [2,
1.5])
robot.waitInterpolationOfGroup('larm')
robot.waitInterpolationOfGroup('rarm')
robot.goInitial()
```
solves your program.

130s, 2017-04-17:
Thank you @k-okada for the correction. I think I was just confused.
I don't think any action is needed for this including updating document.

closed 2017-04-17 by 130s