tork-a/rtmros_nextage #308: checkEncoders gets stuck

130s, 2017-04-05:
```
In [1]: robot.checkEncoders()
[hrpsys.py]  calib-joint all 
[hrpsys.py]  done
Turn on Hand Servo
```
Stuck here and not the command doesn't return.

Full log.
```
$ rospack find hironx_ros_bridge
/opt/ros/indigo/share/hironx_ros_bridge
$ rosversion hironx_ros_bridge
1.1.22

$ ipython -i `rospack find nextage_ros_bridge`/script/nextage.py -- --host nextage --modelfile /opt/jsk/etc/HIRONX/model/main.wrl
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
ROS Master not found running. If you intended to use it (either for rosbridge or for Gazebo purposes, make sure you run necessary nodes.
configuration ORB with nextage:15005
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py]  waitForRTCManagerAndRoboHardware has renamed to waitForRTCManagerAndRoboHardware: Please update your code
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7fd9420c7cf8> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7fd9420c7cf8> isActive? = False 
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7fd9420c7ea8>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x7fd9420c7e60>
[hrpsys.py]    version  =  315.1.8
;;
;; Loading ImpedanceController < 315.1.9
;;
[hrpsys.py]   bodyinfo URL = file:///opt/jsk/etc/HIRONX/model/main.wrl
[hrpsys.py] no hrpsys components found running. creating now
[hrpsys.py]   eval : [self.seq, self.seq_svc, self.seq_version] = self.createComp("SequencePlayer","seq")
[hrpsys.py] create Comp -> SequencePlayer : <hrpsys.rtm.RTcomponent instance at 0x7fd9420c0680> (315.1.8)
[hrpsys.py] create CompSvc -> SequencePlayer Service : <OpenHRP._objref_SequencePlayerService instance at 0x7fd9420d1050>
[hrpsys.py]   eval : [self.sh, self.sh_svc, self.sh_version] = self.createComp("StateHolder","sh")
[hrpsys.py] create Comp -> StateHolder : <hrpsys.rtm.RTcomponent instance at 0x7fd9420b2878> (315.1.8)
[hrpsys.py] create CompSvc -> StateHolder Service : <OpenHRP._objref_StateHolderService instance at 0x7fd9420c87a0>
[hrpsys.py]   eval : [self.fk, self.fk_svc, self.fk_version] = self.createComp("ForwardKinematics","fk")
[hrpsys.py] create Comp -> ForwardKinematics : <hrpsys.rtm.RTcomponent instance at 0x7fd9420d1dd0> (315.1.8)
[hrpsys.py] create CompSvc -> ForwardKinematics Service : <OpenHRP._objref_ForwardKinematicsService instance at 0x7fd9420c0d88>
[hrpsys.py]   eval : [self.ic, self.ic_svc, self.ic_version] = self.createComp("ImpedanceController","ic")
[hrpsys.py] create Comp -> ImpedanceController : <hrpsys.rtm.RTcomponent instance at 0x7fd9420d4830> (315.1.8)
[hrpsys.py] create CompSvc -> ImpedanceController Service : <OpenHRP._objref_ImpedanceControllerService instance at 0x7fd9420c5368>
[hrpsys.py]   eval : [self.el, self.el_svc, self.el_version] = self.createComp("SoftErrorLimiter","el")
[hrpsys.py] create Comp -> SoftErrorLimiter : <hrpsys.rtm.RTcomponent instance at 0x7fd9420d1bd8> (315.1.8)
[hrpsys.py] create CompSvc -> SoftErrorLimiter Service : <OpenHRP._objref_SoftErrorLimiterService instance at 0x7fd9420bf320>
[hrpsys.py]   eval : [self.sc, self.sc_svc, self.sc_version] = self.createComp("ServoController","sc")
[hrpsys.py] create Comp -> ServoController : <hrpsys.rtm.RTcomponent instance at 0x7fd9420d4dd0> (315.1.8)
[hrpsys.py] create CompSvc -> ServoController Service : <OpenHRP._objref_ServoControllerService instance at 0x7fd9420d1680>
[hrpsys.py]   eval : [self.log, self.log_svc, self.log_version] = self.createComp("DataLogger","log")
[hrpsys.py] create Comp -> DataLogger : <hrpsys.rtm.RTcomponent instance at 0x7fd9420c0fc8> (315.1.8)
[hrpsys.py] create CompSvc -> DataLogger Service : <OpenHRP._objref_DataLoggerService instance at 0x7fd9420b2cb0>
[hrpsys.py]   eval : [self.rmfo, self.rmfo_svc, self.rmfo_version] = self.createComp("AbsoluteForceSensor","rmfo")
[hrpsys.py] create Comp -> AbsoluteForceSensor : <hrpsys.rtm.RTcomponent instance at 0x7fd9420d41b8> (315.1.8)
[hrpsys.py] create CompSvc -> AbsoluteForceSensor Service : <OpenHRP._objref_AbsoluteForceSensorService instance at 0x7fd9420c5098>
[hrpsys.py] connecting hrpsys components
[rtm.py]    Connect sh.qOut - ic.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect ic.q - el.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[... 83 lines cut ...]

ROSInitException: No ROS Master found. Without it, you cannot use ROS from this script, but you can still use RTM without issues. To use ROS, do not forget to run rosbridge. How to do so? --> http://wiki.ros.org/rtmros_nextage/Tutorials/Operating%20Hiro%2C%20NEXTAGE%20OPEN

In [1]: robot.checkEncoders()
[hrpsys.py]  calib-joint all 
[hrpsys.py]  done
Turn on Hand Servo
```
Stuck here.

Log from server side:
```
$ more /opt/jsk/var/log/rtcd.log
:
2:stageCalibration[12] = 255
2:stageCalibration[13] = 255
2:stageCalibration[14] = 255
readServoState . done.
Timeover: processing time = 11021.3[ms]
11021.3,  0.0,  0.0,  0.0,  0.0,  0.0,  0.0,  0.0,StateHolder::goActual()
  0.0, 
14:33:22.485103: servo error limit over: joint = RARM_JOINT2, qRef = -71.0155[deg], q = -129.10
7[deg]
[ServoSerial] sending : FA AF 02 00 24 01 01 01 27  - 9
```

130s, 2017-04-05:
Temporary fix on the server-side is to not run `ServoController RTC` https://github.com/start-jsk/rtmros_hironx/issues/480

k-okada, 2017-04-07:
> start-jsk/rtmros_hironx#480

may be you'd better to rever this
https://github.com/tork-a/rtmros_nextage/pull/239

k-okada, 2017-04-21:
Or calling
```
[hrpsys.py] no hrpsys components found running. creating now
```
Is wrong, 
We need to fix this code not going into this node, may be checking all node is necessary, except servo controller or if we have first 3 RTC, we assume everything is installed.
https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L356

130s, 2017-04-21:
I will send a PR next week for this.