tork-a/rtmros_nextage #310: Problem with nextage_ueye_stereo.launch

antgarmun, 2017-04-06:
Hello, I need help.

I am triying to use the Nextage`s stereo vision but It doesn't work. It seems like there is a problem with the ueye_cam package or something similar. I'm not an expert in this kind of things, so I need a step by step tutorial or somethin like this. Anybody can help me please?

130s, 2017-04-06:
Good news: we're willing to help you!

> It doesn't work

No so good news: you basically provided no info for us to help :/ Please see this [support guideline (borrowing from ros.org)](http://wiki.ros.org/Support#Guidelines_for_asking_a_question_.28Please_read_before_posting.29) and give us more info.

antgarmun, 2017-04-06:
Ok, will give you more info.

I am working with the nextage robot . I can move it using ipython comands like nxc.goOffPose(), nxc.goInitial(), etc. But when I try to use the stereo cameras, with the comand "roslaunch nextage_ros_bridge nextage_ueye_stereo.launch" like the Ros Wiki explains (http://wiki.ros.org/rtmros_nextage/Tutorials/Setup%20stereo%20camera), but I have this result:

auto-starting new master
process[master]: started with pid [22947]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f82b33d2-1ab9-11e7-b97a-7c05073c227f
process[rosout-1]: started with pid [22960]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [22972]
[ INFO] [1491477169.849736220]: Initializing nodelet with 8 worker threads.
process[ueye_cam_nodelet_master-3]: started with pid [22996]
process[ueye_cam_nodelet_slave1-4]: started with pid [23012]
process[left_force_subscribe-5]: started with pid [23028]
[ERROR] [1491477169.961185652]: Failed to load nodelet [/ueye_cam_nodelet_master] of type [ueye_cam/ueye_cam_nodelet]: Failed to load library /home/kwduser/ros/hydro/catkin_ws/devel/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.17' not found (required by /usr/lib/libueye_api.so))
[FATAL] [1491477169.961357675]: Service call failed!
process[right_force_subscribe-6]: started with pid [23031]
WARNING: topic [/left/camera_info] does not appear to be published yet
process[right/image_proc-7]: started with pid [23041]
[ERROR] [1491477170.076450188]: Failed to load nodelet [/ueye_cam_nodelet_slave1] of type [ueye_cam/ueye_cam_nodelet]: Failed to load library /home/kwduser/ros/hydro/catkin_ws/devel/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.17' not found (required by /usr/lib/libueye_api.so))
[FATAL] [1491477170.076699667]: Service call failed!
WARNING: topic [/right/camera_info] does not appear to be published yet
process[left/image_proc-8]: started with pid [23068]
process[stereo_image_proc-9]: started with pid [23093]
process[head_link_to_stereo_link-10]: started with pid [23128]
[ueye_cam_nodelet_master-3] process has died [pid 22996, exit code 255, cmd /opt/ros/hydro/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_master __log:=/home/kwduser/.ros/log/f82b33d2-1ab9-11e7-b97a-7c05073c227f/ueye_cam_nodelet_master-3.log].
log file: /home/kwduser/.ros/log/f82b33d2-1ab9-11e7-b97a-7c05073c227f/ueye_cam_nodelet_master-3*.log
process[stereo_link_to_stereo_frame-11]: started with pid [23191]
[ueye_cam_nodelet_slave1-4] process has died [pid 23012, exit code 255, cmd /opt/ros/hydro/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_slave1 __log:=/home/kwduser/.ros/log/f82b33d2-1ab9-11e7-b97a-7c05073c227f/ueye_cam_nodelet_slave1-4.log].
log file: /home/kwduser/.ros/log/f82b33d2-1ab9-11e7-b97a-7c05073c227f/ueye_cam_nodelet_slave1-4*.log
 ...

130s, 2017-04-06:
Thanks for more info. From the error message I get an impression that this has something with the camera driver software that is just called from the launch file in this repo.

Does https://github.com/tork-a/rtmros_nextage/issues/295 help?
If it doesn't, please provide the output of the following. It should be something like `1.0.15-0`.:

```
dpkg -p ros-indigo-ueye-cam | grep Ver
```
If it's older than that then please update and try.

I also noticed that you may be using ROS Hydro, which we NEXTAGE Open maintainers [stopped supporting in 2015](https://github.com/start-jsk/rtmros_hironx/issues/396). That said I'm afraid we may not be able to further assist you if you would keep using the unsupported distro.

antgarmun, 2017-04-07:
Thanks you for your help. I have solved the problem. It was a problem with the libraries installation of IDS cameras.

However, now I have another problem, maybe you can help me:

auto-starting new master
process[master]: started with pid [20280]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2a6c32bc-1b7b-11e7-98cf-7c05073c227f
process[rosout-1]: started with pid [20293]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [20305]
[ INFO] [1491560147.032480837]: Initializing nodelet with 8 worker threads.
process[ueye_cam_nodelet_master-3]: started with pid [20329]
process[ueye_cam_nodelet_slave1-4]: started with pid [20345]
[ INFO] [1491560147.136594223]: Unable to open camera calibration file [/home/kwduser/.ros/camera_info/right.yaml]
process[left_force_subscribe-5]: started with pid [20370]
process[right_force_subscribe-6]: started with pid [20430]
WARNING: topic [/left/camera_info] does not appear to be published yet
process[right/image_proc-7]: started with pid [20450]
WARNING: topic [/right/camera_info] does not appear to be published yet
process[left/image_proc-8]: started with pid [20491]
process[stereo_image_proc-9]: started with pid [20517]
[ERROR] [1491560147.364045649]: Could not open UEye camera ID 1 (IS_CANT_OPEN_DEVICE)
[ERROR] [1491560147.364103770]: Failed to initialize UEye camera 'right'
process[head_link_to_stereo_link-10]: started with pid [20626]
[ INFO] [1491560147.406240696]: Unable to open camera calibration file [/home/kwduser/.ros/camera_info/left.yaml]
process[stereo_link_to_stereo_frame-11]: started with pid [20747]
subscribed to [/right/camera_info]
subscribed to [/left/camera_info]
[ERROR] [1491560148.163974991]: Could not open UEye camera ID 2 (IS_IO_REQUEST_FAILED)
[ERROR] [1491560148.164022507]: Failed to initialize UEye camera 'left'

The launch file can't open the cameras, something similar happens with the ueyedemo software, that gives me the following message: "Open camera failed (Code: 2)"

Do you know what is the problem?

Thanks you so much for the help.

130s, 2017-04-08:
We may need this https://github.com/ros-planning/moveit_resources/pull/14 from the released deb. I'll make a release there (even though why that's not the issue on Indigo and Jade isn't known to me).

130s, 2017-04-08:
@antgarmun I posted a wrong comment earlier sorry. Anyways,

> ```
> [ INFO] [1491560147.136594223]: Unable to open camera calibration file [/home/kwduser/.ros/camera_info/right.yaml]
> ```

Necessary yaml file is not found. Have you done [calibration as noted in the tutorial](http://wiki.ros.org/rtmros_nextage/Tutorials/Setup%20stereo%20camera#Calibrate_stereo_camera)?

antgarmun, 2017-04-10:
@130s When I try to run the calibration process, I get this error:

"Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying."

I know that the master is running because I can see the cameras en the IDS Camera Manager, however I can't see the cameras' images with this software because it returns an error: "Open camera failed (Code: 2)"

130s, 2017-04-10:
> I know that the master is running because I can see the cameras en the IDS Camera Manager, 

If I remember correctly, `IDS Camera Manager` is a proprietary software that ROS package `ueye_cam` doesn't depend on. So whether ROS master is running or not is not known yet to me from the information above only.

> however I can't see the cameras' images with this software because it returns an error: "Open camera failed (Code: 2)"

Could you share the **entire** output starting from `roslaunch ...` line? It's a good tip when you share the output of roslaunch. It's a bit hard to get a whole picture of what's happening with the lines deliberately selected.

Also please let us know if you're using the Ubuntu that came with the robot or your own computer. `nextage_ueye_stereo.launch` expects the cameras are connected to the computer where you run this launch file.

antgarmun, 2017-04-10:
I am using the Ubuntu that came with the robot, with all connected correctly, I run the "roslaunch nextage_ros_bridge nextage_ueye_stereo.launch"  comand and I get this output:

kwduser@hiroconsole:~$ roslaunch nextage_ros_bridge nextage_ueye_stereo.launch 
... logging to /home/kwduser/.ros/log/83b9afe2-1dd9-11e7-ad82-7c05073c227f/roslaunch-hiroconsole-5607.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/kwduser/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://hiroconsole:52617/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion
 * /ueye_cam_nodelet_master/auto_exposure
 * /ueye_cam_nodelet_master/auto_frame_rate
 * /ueye_cam_nodelet_master/auto_gain
 * /ueye_cam_nodelet_master/auto_white_balance
 * /ueye_cam_nodelet_master/binning
 * /ueye_cam_nodelet_master/blue_gain
 * /ueye_cam_nodelet_master/camera_id
 * /ueye_cam_nodelet_master/camera_intrinsics_file
 * /ueye_cam_nodelet_master/camera_name
 * /ueye_cam_nodelet_master/camera_parameters_file
 * /ueye_cam_nodelet_master/camera_topic
 * /ueye_cam_nodelet_master/color_mode
 * /ueye_cam_nodelet_master/exposure
 * /ueye_cam_nodelet_master/ext_trigger_mode
 * /ueye_cam_nodelet_master/flash_delay
 * /ueye_cam_nodelet_master/flash_duration
 * /ueye_cam_nodelet_master/frame_rate
 * /ueye_cam_nodelet_master/gain_boost
 * /ueye_cam_nodelet_master/green_gain
 * /ueye_cam_nodelet_master/image_height
 * /ueye_cam_nodelet_master/image_left
 * /ueye_cam_nodelet_master/image_top
 * /ueye_cam_nodelet_master/image_width
 * /ueye_cam_nodelet_master/master_gain
 * /ueye_cam_nodelet_master/pixel_clock
 * /ueye_cam_nodelet_master/red_gain
 * /ueye_cam_nodelet_master/sensor_scaling
 * /ueye_cam_nodelet_master/subsampling
 * /ueye_cam_nodelet_master/white_balance_blue_offset
 * /ueye_cam_nodelet_master/white_balance_red_offset
 * /ueye_cam_nodelet_slave1/auto_exposure
 * /ueye_cam_nodelet_slave1/auto_frame_rate
 * /ueye_cam_nodelet_slave1/auto_gain
 * /ueye_cam_nodelet_slave1/auto_white_balance
 * /ueye_cam_nodelet_slave1/binning
 * /ueye_cam_nodelet_slave1/blue_gain
 * /ueye_cam_nodelet_slave1/camera_id
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file
 * /ueye_cam_nodelet_slave1/camera_name
 * /ueye_cam_nodelet_slave1/camera_parameters_file
 * /ueye_cam_nodelet_slave1/camera_topic
 * /ueye_cam_nodelet_slave1/color_mode
 * /ueye_cam_nodelet_slave1/exposure
 * /ueye_cam_nodelet_slave1/ext_trigger_mode
 * /ueye_cam_nodelet_slave1/flash_delay
 * /ueye_cam_nodelet_slave1/flash_duration
 * /ueye_cam_nodelet_slave1/frame_rate
 * /ueye_cam_nodelet_slave1/gain_boost
 * /ueye_cam_nodelet_slave1/green_gain
 * /ueye_cam_nodelet_slave1/image_height
 * /ueye_cam_nodelet_slave1/image_left
 * /ueye_cam_nodelet_slave1/image_top
 * /ueye_cam_nodelet_slave1/image_width
 * /ueye_cam_nodelet_slave1/master_gain
 * /ueye_cam_nodelet_slave1/pixel_clock
 * /ueye_cam_nodelet_slave1/red_gain
 * /ueye_cam_nodelet_slave1/sensor_scaling
 * /ueye_cam_nodelet_slave1/subsampling
 * /ueye_cam_nodelet_slave1/white_balance_blue_offset
 * /ueye_cam_nodelet_slave1/white_balance_red_offset

NODES
  /left/
    image_proc (image_proc/image_proc)
  /right/
    image_proc (image_proc/image_proc)
  /
    head_link_to_stereo_link (tf/static_transform_publisher)
    left_force_subscribe (rostopic/rostopic)
    nodelet_manager (nodelet/nodelet)
    right_force_subscribe (rostopic/rostopic)
    stereo_image_proc (stereo_image_proc/stereo_image_proc)
    stereo_link_to_stereo_frame (tf/static_transform_publisher)
    ueye_cam_nodelet_master (nodelet/nodelet)
    ueye_cam_nodelet_slave1 (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [5621]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 83b9afe2-1dd9-11e7-ad82-7c05073c227f
process[rosout-1]: started with pid [5634]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [5646]
[ INFO] [1491820571.926025274]: Initializing nodelet with 8 worker threads.
process[ueye_cam_nodelet_master-3]: started with pid [5670]
process[ueye_cam_nodelet_slave1-4]: started with pid [5686]
[ INFO] [1491820572.045746118]: Unable to open camera calibration file [/home/kwduser/.ros/camera_info/right.yaml]
process[left_force_subscribe-5]: started with pid [5768]
process[right_force_subscribe-6]: started with pid [5786]
WARNING: topic [/left/camera_info] does not appear to be published yet
process[right/image_proc-7]: started with pid [5806]
WARNING: topic [/right/camera_info] does not appear to be published yet
process[left/image_proc-8]: started with pid [5835]
[ERROR] [1491820572.257116761]: Could not open UEye camera ID 1 (IS_CANT_OPEN_DEVICE)
[ERROR] [1491820572.257186873]: Failed to initialize UEye camera 'right'
process[stereo_image_proc-9]: started with pid [5954]
[ INFO] [1491820572.308924659]: Unable to open camera calibration file [/home/kwduser/.ros/camera_info/left.yaml]
process[head_link_to_stereo_link-10]: started with pid [6069]
process[stereo_link_to_stereo_frame-11]: started with pid [6142]
subscribed to [/right/camera_info]
subscribed to [/left/camera_info]
[ERROR] [1491820573.077522611]: Could not open UEye camera ID 2 (IS_IO_REQUEST_FAILED)
[ERROR] [1491820573.077570604]: Failed to initialize UEye camera 'left'

antgarmun, 2017-04-11:
I have solved this problem but I have another now. I get this output:

kwduser@hiroconsole:~$ roslaunch nextage_ros_bridge nextage_ueye_stereo.launch
... logging to /home/kwduser/.ros/log/47f3ac4a-1e80-11e7-ba7a-7c05073c227f/roslaunch-hiroconsole-12487.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/kwduser/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://hiroconsole:41322/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion
 * /ueye_cam_nodelet_master/auto_exposure
 * /ueye_cam_nodelet_master/auto_frame_rate
 * /ueye_cam_nodelet_master/auto_gain
 * /ueye_cam_nodelet_master/auto_white_balance
 * /ueye_cam_nodelet_master/binning
 * /ueye_cam_nodelet_master/blue_gain
 * /ueye_cam_nodelet_master/camera_id
 * /ueye_cam_nodelet_master/camera_intrinsics_file
 * /ueye_cam_nodelet_master/camera_name
 * /ueye_cam_nodelet_master/camera_parameters_file
 * /ueye_cam_nodelet_master/camera_topic
 * /ueye_cam_nodelet_master/color_mode
 * /ueye_cam_nodelet_master/exposure
 * /ueye_cam_nodelet_master/ext_trigger_mode
 * /ueye_cam_nodelet_master/flash_delay
 * /ueye_cam_nodelet_master/flash_duration
 * /ueye_cam_nodelet_master/frame_rate
 * /ueye_cam_nodelet_master/gain_boost
 * /ueye_cam_nodelet_master/green_gain
 * /ueye_cam_nodelet_master/image_height
 * /ueye_cam_nodelet_master/image_left
 * /ueye_cam_nodelet_master/image_top
 * /ueye_cam_nodelet_master/image_width
 * /ueye_cam_nodelet_master/master_gain
 * /ueye_cam_nodelet_master/pixel_clock
 * /ueye_cam_nodelet_master/red_gain
 * /ueye_cam_nodelet_master/sensor_scaling
 * /ueye_cam_nodelet_master/subsampling
 * /ueye_cam_nodelet_master/white_balance_blue_offset
 * /ueye_cam_nodelet_master/white_balance_red_offset
 * /ueye_cam_nodelet_slave1/auto_exposure
 * /ueye_cam_nodelet_slave1/auto_frame_rate
 * /ueye_cam_nodelet_slave1/auto_gain
 * /ueye_cam_nodelet_slave1/auto_white_balance
 * /ueye_cam_nodelet_slave1/binning
 * /ueye_cam_nodelet_slave1/blue_gain
 * /ueye_cam_nodelet_slave1/camera_id
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file
 * /ueye_cam_nodelet_slave1/camera_name
 * /ueye_cam_nodelet_slave1/camera_parameters_file
 * /ueye_cam_nodelet_slave1/camera_topic
 * /ueye_cam_nodelet_slave1/color_mode
 * /ueye_cam_nodelet_slave1/exposure
 * /ueye_cam_nodelet_slave1/ext_trigger_mode
 * /ueye_cam_nodelet_slave1/flash_delay
 * /ueye_cam_nodelet_slave1/flash_duration
 * /ueye_cam_nodelet_slave1/frame_rate
 * /ueye_cam_nodelet_slave1/gain_boost
 * /ueye_cam_nodelet_slave1/green_gain
 * /ueye_cam_nodelet_slave1/image_height
 * /ueye_cam_nodelet_slave1/image_left
 * /ueye_cam_nodelet_slave1/image_top
 * /ueye_cam_nodelet_slave1/image_width
 * /ueye_cam_nodelet_slave1/master_gain
 * /ueye_cam_nodelet_slave1/pixel_clock
 * /ueye_cam_nodelet_slave1/red_gain
 * /ueye_cam_nodelet_slave1/sensor_scaling
 * /ueye_cam_nodelet_slave1/subsampling
 * /ueye_cam_nodelet_slave1/white_balance_blue_offset
 * /ueye_cam_nodelet_slave1/white_balance_red_offset

NODES
  /left/
    image_proc (image_proc/image_proc)
  /right/
    image_proc (image_proc/image_proc)
  /
    head_link_to_stereo_link (tf/static_transform_publisher)
    left_force_subscribe (rostopic/rostopic)
    nodelet_manager (nodelet/nodelet)
    right_force_subscribe (rostopic/rostopic)
    stereo_image_proc (stereo_image_proc/stereo_image_proc)
    stereo_link_to_stereo_frame (tf/static_transform_publisher)
    ueye_cam_nodelet_master (nodelet/nodelet)
    ueye_cam_nodelet_slave1 (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12501]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 47f3ac4a-1e80-11e7-ba7a-7c05073c227f
process[rosout-1]: started with pid [12514]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [12526]
[ INFO] [1491892197.611709613]: Initializing nodelet with 8 worker threads.
process[ueye_cam_nodelet_master-3]: started with pid [12550]
process[ueye_cam_nodelet_slave1-4]: started with pid [12566]
[ INFO] [1491892197.707943315]: Unable to open camera calibration file [/home/kwduser/.ros/camera_info/right.yaml]
process[left_force_subscribe-5]: started with pid [12650]
process[right_force_subscribe-6]: started with pid [12666]
WARNING: topic [/left/camera_info] does not appear to be published yet
process[right/image_proc-7]: started with pid [12686]
WARNING: topic [/right/camera_info] does not appear to be published yet
process[left/image_proc-8]: started with pid [12719]
process[stereo_image_proc-9]: started with pid [12820]
process[head_link_to_stereo_link-10]: started with pid [12916]
process[stereo_link_to_stereo_frame-11]: started with pid [12962]
subscribed to [/right/camera_info]
[FATAL] [1491892198.270392097]: Service call failed!
[FATAL] [1491892198.270506781]: Service call failed!
[nodelet_manager-2] process has died [pid 12526, exit code -11, cmd /opt/ros/hydro/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/kwduser/.ros/log/47f3ac4a-1e80-11e7-ba7a-7c05073c227f/nodelet_manager-2.log].
log file: /home/kwduser/.ros/log/47f3ac4a-1e80-11e7-ba7a-7c05073c227f/nodelet_manager-2*.log
[ueye_cam_nodelet_slave1-4] process has died [pid 12566, exit code 255, cmd /opt/ros/hydro/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_slave1 __log:=/home/kwduser/.ros/log/47f3ac4a-1e80-11e7-ba7a-7c05073c227f/ueye_cam_nodelet_slave1-4.log].
log file: /home/kwduser/.ros/log/47f3ac4a-1e80-11e7-ba7a-7c05073c227f/ueye_cam_nodelet_slave1-4*.log
[ueye_cam_nodelet_master-3] process has died [pid 12550, exit code 255, cmd /opt/ros/hydro/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_master __log:=/home/kwduser/.ros/log/47f3ac4a-1e80-11e7-ba7a-7c05073c227f/ueye_cam_nodelet_master-3.log].
log file: /home/kwduser/.ros/log/47f3ac4a-1e80-11e7-ba7a-7c05073c227f/ueye_cam_nodelet_master-3*.log

130s, 2017-04-11:
It seems to me that you're getting the same error from earlier https://github.com/tork-a/rtmros_nextage/issues/310#issuecomment-292500449 and I posted a comment about it https://github.com/tork-a/rtmros_nextage/issues/310#issuecomment-292694162
Although the error message is labeled as `[INFO]` instead of something more critical, I think that file must exist in order to proceed.

antgarmun, 2017-04-12:
I don't know what is the problem, but when I run the "rostopic list" command, I get this output:
/rosout
/rosout_agg

The cameras doesn't appear, so I think that the problem comes from the integration of the cameras in ROS.

antgarmun, 2017-04-12:
@130s I have solved the problem running the comand "catkin_make" again inside the workspace. After that, the stereo vision works correctly. 

Thanks you so much for your help.

closed 2017-04-12 by antgarmun

130s, 2017-04-17:
> I have solved the problem running the comand "catkin_make" again inside the workspace. After that, the stereo vision works correctly.

I don't know what was happening then. I assume you were trying to use [ueye_cam](http://wiki.ros.org/ueye_cam) package from source, not deb binary, then you sourced `setup.*sh` but didn't compile it.