tork-a/rtmros_nextage #313: using moveit ik on #312

k-okada, 2017-04-23:
#312 calls setTargetPose and move robot arm to get current list of angle values, and then, call playPatter for waypoint control, we can use MoveIt's IK service ( http://answers.ros.org/question/82151/how-can-i-move-my-arm-to-a-specified-point-in-space-using-moveit/), instead.

y-yosuke, 2017-05-01:
I wrote a example of waypoints motion using MoveIt! .

```
#!/usr/bin/env python

import rospy
import geometry_msgs.msg
import copy
import tf

from moveit_commander import MoveGroupCommander

import nextage_rtm_playpattern as nxtpp

def setTargetPoseSequenceMoveIt(limb, pos_list, rpy_list, tm_list):
    wpose = geometry_msgs.msg.Pose()
    waypoints = []
    
    for pos, rpy, tm in zip(pos_list, rpy_list, tm_list):
        quaternion = tf.transformations.quaternion_from_euler(rpy[0], rpy[1], rpy[2])
        wpose.position.x = pos[0]
        wpose.position.y = pos[1]
        wpose.position.z = pos[2]
        wpose.orientation.x = quaternion[0]
        wpose.orientation.y = quaternion[1]
        wpose.orientation.z = quaternion[2]
        wpose.orientation.w = quaternion[3]
        waypoints.append(copy.deepcopy(wpose))
    
    (pln, frc) = limb.compute_cartesian_path(
                             waypoints,   # waypoints to follow
                             0.01,        # eef_step
                             0.0)         # jump_threshold
    
    return pln, frc

if __name__ == '__main__':
    rospy.init_node('commander_example', anonymous=True)

    limb_name = "left_arm"  # Limb 'right_arm', 'left_arm'
    group = MoveGroupCommander(limb_name)
  
    rarm_initial_pose = group.get_current_pose().pose
    rospy.loginfo("Got Initial Pose \n{}".format(rarm_initial_pose))
    
    pose_target = geometry_msgs.msg.Pose()
    pose_target = copy.deepcopy(rarm_initial_pose)
    
    positions_arm = []  # All elements are lists. E.g. pos1 = [x1, y1, z1]
    rpys_arm = []       # All elements are lists. E.g. rpy1 = [r1, p1, y1]
    time_list = []      # Time list [t1, t2, t3,...]
    
    # Rectangular Positions, RPYs, Time List
    rect_xyzs = nxtpp.rectangularPositions(dp_a=[0.25, 0.0, 0.1], dp_b=[0.45, 0.2, 0.1])
    rect_rpys = nxtpp.samePostureRPY([-3.073437, -1.569023, 3.073247], len(rect_xyzs))
    rect_time = nxtpp.equalTimeList(10.0, len(rect_xyzs))
    
    # Circular Positions, RPYs, Time List
    circ_xyzs = nxtpp.circularPositions(center=[0.35, 0.1, 0.1], radius=0.1 ,steps=12)
    circ_rpys = nxtpp.samePostureRPY([-3.073437, -1.569023, 3.073247], len(circ_xyzs))
    circ_time = nxtpp.equalTimeList(10.0, len(circ_xyzs))

    # Adjust Transition Time
    rect_time[0] += 1.0
    circ_time[0] += 1.0
    
    # Combine Patterns
    positions_arm = rect_xyzs + circ_xyzs
    rpys_arm      = rect_rpys + circ_rpys
    time_list     = rect_time + circ_time

    print('Limb: %s' % limb_name)
    print('Positions (%d): %s' % (len(positions_arm), positions_arm))
    print('RPY Pattern (%d): %s' % (len(rpys_arm), rpys_arm))
    print('Time List (%d): %s' % (len(time_list), time_list))    
    
    # plan waypoints
    rospy.loginfo("Planing Waypoints")
    plan, flaction = setTargetPoseSequenceMoveIt(group, positions_arm, rpys_arm, time_list)
    
    # wait for playing the plan on MoveIt
    rospy.loginfo("Waiting while the plan is visualized on RViz")
    rospy.sleep(20)
    
    # execute the plan
    rospy.loginfo("Execute!!")
    group.execute(plan)
    rospy.loginfo("Executiion Done")
    rospy.sleep(5)

    # move to the initial pose
    rospy.loginfo("Move to Initial Pose")
    pose_target = copy.deepcopy(rarm_initial_pose)
    rospy.loginfo("set target to {}".format(pose_target))
    group.set_pose_target(pose_target)
    plan = group.plan()
    ret = group.go()

```

y-yosuke, 2017-05-04:
I wrote descriptions of "Using MoveIt!" in ROS Wiki.
http://wiki.ros.org/rtmros_nextage/Tutorials/Task-oriented%20programming#Using_MoveIt.21

k-okada, 2017-05-06:
thank you, looks good to me.

closed 2017-05-06 by k-okada