tork-a/rtmros_nextage #326: "robot.ic_svc.getImpedanceControllerParam" does not return what was passed by startImpedance

130s, 2017-05-11:
The values that `getImpedanceControllerParam` returns are not the same with what are passed to `startImpedance`.

So far I've only tested this on simulator. On real robot will be tested very soon.

Not sure how this relates or not to #325.

```
term-1$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch

term-2$ $ rospack find nextage_ros_bridge 
/opt/ros/indigo/share/nextage_ros_bridge
$ dpkg -p ros-indigo-nextage-ros-bridge |grep Ver
Version: 0.7.16-0trusty-20170509-121945-0800
$ ipython -i `rospack find nextage_ros_bridge`/script/nextage.py
:
In [1]: robot.startImpedance('larm', M_p = 15.0, D_p = 15.0, K_p = 15.0, M_r = 15.0, D_r = 300.0, K_r = 30                                                                                                   [55/510]
0.0)
[hrpsys.py] , Failt to getImpedanceControllerParam(larm)
startImpedance  call is done. This does't necessarily mean the function call was successful, since not all methods internally called return status

In [2]: robot.ic_svc.getImpedanceControllerParam('larm')                                                  
Out[2]: 
(False,
 OpenHRP.ImpedanceControllerService.impedanceParam(M_p=10.0, D_p=200.0, K_p=400.0, M_r=5.0, D_r=100.0, K_r=200.0, force_gain=[1.0, 1.0, 1.0], moment_gain=[1.0, 1.0, 1.0], sr_gain=1.0, avoid_gain=0.001, reference_g
ain=0.01, manipulability_limit=0.1, controller_mode=MODE_IDLE, ik_optional_weight_vector=[], use_sh_base_pos_rpy=False))

$ rtls localhost:15005/        
StateHolderServiceROSBridge.rtc        fk.rtc
DataLoggerServiceROSBridge.rtc         longfloor(Robot)0.rtc
ImageSensorROSBridge_HandLeft.rtc      sh.rtc
HiroNX(Robot)0.rtc                     ImageSensorROSBridge_HeadRight.rtc
seq.rtc                                HrpsysJointTrajectoryBridge0.rtc
log.rtc                                HGcontroller0.rtc
*sc.rtc                                ModelLoader
el.rtc                                 CollisionDetector0.rtc
ImageSensorROSBridge_HandRight.rtc     ImageSensorROSBridge_HeadLeft.rtc
HrpsysSeqStateROSBridge0.rtc           SequencePlayerServiceROSBridge.rtc
ForwardKinematicsServiceROSBridge.rtc  tork-kudu1.host_cxt/
ic.rtc                                 *rmfo.rtc
$ rosnode list
/DataLoggerServiceROSBridge
/ForwardKinematicsServiceROSBridge
/HrpsysJointTrajectoryBridge
/HrpsysSeqStateROSBridge
/SequencePlayerServiceROSBridge
/StateHolderServiceROSBridge
/collision_state
/diagnostic_aggregator
/hand_left/ImageSensorROSBridge_HandLeft
/hand_right/ImageSensorROSBridge_HandRight
/hironx_ros_client
/hrpsys_profile
/hrpsys_py
/hrpsys_ros_diagnostics
/hrpsys_state_publisher
/left/ImageSensorROSBridge_HeadLeft
/moveit_python_wrappers_1494531264696351424
/moveit_python_wrappers_1494531278372584200
/right/ImageSensorROSBridge_HeadRight
/rosout
/rqt_hironxo
$ rostopic list|grep sensor
$ rostopic list|grep lh
```

130s, 2017-05-11:
In [HIRONX.startImpedance_315_3](https://github.com/start-jsk/rtmros_hironx/blob/02c82654802336ae7deba01b65b564a83fcf2f61/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L1004), the method returns when `ic_svc.getImpedanceControllerParam` returns `False` (which I think means no ImpedanceControllerParam was found on RTC side).

Is it the intended behavior?

k-okada, 2017-05-12:
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceController.cpp#L789-L795

i think you haven't set config for RTC, please check output of rtcd