tork-a/rtmros_nextage #333: "wait for  HrpsysSeqStateROSBridge0  :  None" continues

130s, 2017-05-22:
On a user's Ubuntu host, `HrpsysSeqStateROSBridge0` can't be connected, both with their real robot and on simulation.

```
term-1$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl

term-2$ rostopic list|grep sensor
term-2$ 
```

```
$ rosversion nextage_ros_bridge
0.7.16
$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=nextage MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
:
^[[0m[ INFO] [1495432036.430229043]: [HrpsysSeqStateROSBridge] @Initilize name : HrpsysSeqSta\
teROSBridge0^[[0m
Loading body customizer "/opt/ros/indigo/share/OpenHRP-3.1/customizer/libSampleCustomizer.so"\
 for springJoint
Loading body customizer "/opt/ros/indigo/share/OpenHRP-3.1/customizer/libSampleBushCustomizer\
.so" for
duplicated sensor Id is specified(id = 0, name = CAMERA_HEAD_R)
duplicated sensor Id is specified(id = 1, name = CAMERA_HEAD_L)
duplicated sensor Id is specified(id = 0, name = rhsensor)
duplicated sensor Id is specified(id = 1, name = lhsensor)
0 physical force sensor : rhsensor
1 physical force sensor : lhsensor
:
[ INFO] [1495434563.794082857]: [HrpsysJointTrajectoryBridge] @Initilize name : HrpsysJointTrajectoryBridge0 done
[WARN] [WallTime: 1495434563.800814] Could not found CollisionDetector(CollisionDetector0), waiting...
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /nextage:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   4 /30
[WARN] [WallTime: 1495434564.868632] Could not found CollisionDetector(CollisionDetector0), waiting...
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc:qCurrent   4 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[WARN] [WallTime: 1495434565.954951] Could not found CollisionDetector(CollisionDetector0), waiting...
[rtmlaunch] Wait for  /nextage:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   5 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc:qCurrent   5 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[WARN] [WallTime: 1495434567.041006] Could not found CollisionDetector(CollisionDetector0), waiting...
[rtmlaunch] Wait for  /nextage:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   6 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc:qCurrent   6 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[WARN] [WallTime: 1495434568.131910] Could not found CollisionDetector(CollisionDetector0), waiting...
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /nextage:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   7 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc:qCurrent   7 /30
[WARN] [WallTime: 1495434569.229564] Could not found CollisionDetector(CollisionDetector0), waiting...
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[ERROR] [WallTime: 1495434570.231523] Could not found CollisionDetector, exiting...
[rtmlaunch] Wait for  /nextage:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   8 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc:qCurrent   8 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[collision_state-18] process has finished cleanly
log file: /home/nxo/.ros/log/f9376922-3eb7-11e7-84f4-000bab77d7ad/collision_state-18*.log
[rtmlaunch] Wait for  /nextage:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   9 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc:qCurrent   9 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
:

$ rosversion nextage_ros_bridge
0.7.16
$ rosversion hironx_ros_bridge
1.1.23
```

Downgrading `*_ros_bridge (source build) didn't help.
```
$ rosversion nextage_ros_bridge
0.7.14
[... 522 lines cut ...]
[rtmlaunch_hrpsys_ros_bridge-12] killing on exit
[sensor_ros_bridge_connect-11] killing on exit
[hrpsys_profile-10] killing on exit
[diagnostic_aggregator-9] killing on exit
[hrpsys_ros_diagnostics-8] killing on exit
[hrpsys_state_publisher-7] killing on exit
[HrpsysJointTrajectoryBridge-6] killing on exit
[HrpsysSeqStateROSBridge-5] killing on exit
[hrpsys_py-4] killing on exit
[hrpsys-3] killing on exit
[modelloader-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
[1mdone[0m
[34m[rtmlaunch] terminate omniNames at port 15005 [0m
```

Reinstalling some DEB files didn't help.
```
$ sudo apt-get purge ros-indigo-rtmros-hironx ros-indigo-rtmros-nextage
$ sudo apt-get autoremove
$ sudo apt-get install ros-indigo-rtmros-hironx ros-indigo-rtmros-nextage
```

130s, 2017-05-23:
Btw, with [a commit](https://github.com/tork-a/rtmros_nextage/pull/305/commits/f942a0bce223c27ed68ce057b3ac851e5c825c39) in the pull request #305 I can reproduce the issue on the simulator on my computer.
```
term-1$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch USE_IMPEDANCE:=true
[rtmlaunch] Start omniNames at port 15005

Mon May 22 22:22:20 2017:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000d0000003139322e3136382e302e313800009d3a0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000000cc7235901007253
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/rosnoodle/.ros/log/cb7c51b2-3f77-11e7-9aad-80fa5b08d1d8/roslaunch-tork-kudu1-29261.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tork-kudu1:43816/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path:
 * /diagnostic_aggregator/pub_rate: 1.0
 * /hand_left/ImageSensorROSBridge_HandLeft/frame_id: /CAMERA_HAND_L
 * /hand_right/ImageSensorROSBridge_HandRight/frame_id: /CAMERA_HAND_R
 * /left/ImageSensorROSBridge_HeadLeft/frame_id: /CAMERA_HEAD_L
 * /right/ImageSensorROSBridge_HeadRight/frame_id: /CAMERA_HEAD_R
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtmnameserver_port: 15005
 * /rtmnameserver_robotname: HiroNX(Robot)0
 * /use_sim_time: True

NODES
  /left/
    ImageSensorROSBridge_HeadLeft (hrpsys_ros_bridge/ImageSensorROSBridge)
  /hand_left/
    ImageSensorROSBridge_HandLeft (hrpsys_ros_bridge/ImageSensorROSBridge)
  /right/
    ImageSensorROSBridge_HeadRight (hrpsys_ros_bridge/ImageSensorROSBridge)
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    ImpedanceControllerServiceROSBridge (hrpsys_ros_bridge/ImpedanceControllerServiceROSBridgeComp)
    RemoveForceSensorLinkOffsetServiceROSBridge (hrpsys_ros_bridge/RemoveForceSensorLinkOffsetServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
[... 803 lines cut ...]
[hrpsys.py] setup joint groups for hrpsys controller
[hrpsys.py] initialized successfully
[ WARN] [1495516960.485587998]: COLLADA warning: The DOM was unable to create an element named copyright at line 10. Probably a schema violation.

[ERROR] [1495516960.594956365]: Target Node visual1/node_joint0_axis0 NOT found!!!

[ WARN] [1495516960.599806400]: could not find binding for axis: kmodel1/jointsid1000/axis0, axis0

[ WARN] [1495516960.603235132]: could not find binding for axis: kmodel1/jointsid1000/axis0, axis0

[ WARN] [1495516960.693611473]: could not find openrave joint WAIST!

[ERROR] [1495516960.700367751]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
[ INFO] [1495516960.700458677]: Loading robot model 'HiroNX'...
[ INFO] [1495516960.700484096]: No root/virtual joint specified in SRDF. Assuming fixed joint
^C[INFO] [WallTime: 1495516969.288101] [0.000000] Joint names; Torso: ['CHEST_JOINT0']
        Head: ['HEAD_JOINT0', 'HEAD_JOINT1']
        L-Arm: ['LARM_JOINT0', 'LARM_JOINT1', 'LARM_JOINT2', 'LARM_JOINT3', 'LARM_JOINT4', 'LARM_JOINT5']
        R-Arm: ['RARM_JOINT0', 'RARM_JOINT1', 'RARM_JOINT2', 'RARM_JOINT3', 'RARM_JOINT4', 'RARM_JOINT5']
[WARN] [WallTime: 1495516969.290552] [0.000000] Moveit is not started yet, if you want to use MoveIt!
Make sure you've launched MoveGroup (e.g. by launching moveit_planning_execution.launch)

In [1]:
Do you really want to exit ([y]/n)?
```

130s, 2017-05-23:
> On a user's Ubuntu host, `HrpsysSeqStateROSBridge0` can't be connected, both with their real robot and on simulation.

I was wrong; on the simulation on this user's Ubuntu computer, this issue is NOT occurring. 
As in https://github.com/tork-a/rtmros_nextage/issues/333#issuecomment-303294270, the simulation on my own computer this issue happens.

k-okada, 2017-05-24:
please confirm that impedance controller is running on simulator with hironx robot with https://github.com/start-jsk/rtmros_hironx/pull/338/files#diff-197cb5094a9e431bf96e62d767605f2fR783, before you try to run them on nextage robot, they are slightly different.

k-okada, 2017-05-24:
> rtmros_hironx/pull/338/files#diff-197cb5094a9e431bf96e62d767605f2fR783,

```
[hrpsys.py] wait for HiroNX(Robot)0 : <hrpsys.rtm.RTcomponent instance at
0x7f7130a59fc8> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance
at 0x7f7130a59fc8> isActive? = True
```
on real robot, it should connect to RobotHardware, not HiroNX(Robot)0.
please check code around
https://github.com/tork-a/rtmros_nextage/blob/indigo-devel/nextage_ros_bridge/script/nextage.py#L69

130s, 2017-05-26:
On simulator on my computer,

> please confirm that impedance controller is running on simulator with hironx robot with https://github.com/start-jsk/rtmros_hironx/pull/338/files#diff-197cb5094a9e431bf96e62d767605f2fR783, before you try to run them on nextage robot, they are slightly different.

Yes Impedance Controller works for Hironx sim.

On the real robot,

> on real robot, it should connect to RobotHardware, not HiroNX(Robot)0.

Good point, I failed to include the full log in my post from the real robot https://github.com/tork-a/rtmros_nextage/issues/333#issue-230507214. But anyway it used the latest deb at that time `hironx_ros_bridge` 1.1.23 that sets [--robot RobotHardware](https://github.com/start-jsk/rtmros_hironx/blob/21b09970f4fd79b891e565bbe30244378413dfc5/hironx_ros_bridge/launch/hironx_ros_bridge_real.launch#L13).

Still, I understand that it's a bit confusing to discuss an issue on the real and simulator in a single ticket even for the same error. On this ticket let us only talk about the real robot issue.

k-okada, 2017-05-26:
> Good point, I failed to include the full log in my post from the real robot #333 (comment). But anyway it used the latest deb at that time hironx_ros_bridge 1.1.23 that sets --robot RobotHardware.

https://github.com/tork-a/rtmros_nextage/issues/333#issue-230507214 logs seems rtcd is dieing , we need rtcd.log and full log of roslaunch output, also you'd better to start with `roslauch -s` when you debug.

130s, 2017-06-27:
Turned out that `sensorId` in `.wrl` needs to be uniquely sequenced PER the types of sensors. E.g. in [openhrp3/sample/model/sample1.wrl](https://github.com/fkanehiro/openhrp3/blob/6883f10630225d637fb04802a4e9feb0c6521966/sample/model/sample1.wrl#L407), for both `AccelerationSensor` and `Gyro` sensorId starts from 0.

closed 2017-06-27 by 130s