tork-a/rtmros_nextage #341: IKFast plugin: Minor fix to apply

ysolana, 2017-06-27:
Hi all,

I am reporting a minor bug about the harmonize function inside the two next files (line 482, indigo-devel branch):
- rtmros_nextage/nextage_ik_plugin/src/nextage_left_arm_ikfast_moveit_plugin.cpp 
- rtmros_nextage/nextage_ik_plugin/src/nextage_right_arm_ikfast_moveit_plugin.cpp

A minus sign is missing in order to properly bound the angles in [-2*PI; 2*PI]
line 482:    while(solution[i] < **-**2*M_PI) { ...

Regards,
Yoann.

130s, 2017-06-27:
Thanks for the headsup. A few questions:
- Have you confirmed the suggested change works? On simulator or on the real robot? 
- Do you think you can open a pull request?

130s, 2017-06-27:
Either way I'd like to ask @k-okada for taking care of this.

ysolana, 2017-06-27:
Hi 130s,

It was tested with the simulator and it works.
I have created the pull request: #342 

Regards,
Yoann.

130s, 2017-06-27:
Thanks!
Continued in #342

closed 2017-06-27 by 130s