tork-a/rtmros_nextage #343: "omniORB.BAD_OPERATION_UnRecognisedOperationName" error upon calling readDigitalOutput

130s, 2017-06-27:
Reported at https://github.com/tork-a/rtmros_nextage/issues/340#issuecomment-311367035 from @antgarmun:

```
In [2]: b=robot.readDigitalOutput()
BAD_OPERATION                             Traceback (most recent call last)
/home/kwduser/<ipython-input-2-4ab7a4ba7bdd> in <module>()
----> 1 b=robot.readDigitalOutput()

/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.pyc in readDigitalOutput(self)
   1554         @return list of int: List of the values in digital input register. Range: 0 or 1.
   1555         '''
-> 1556         ret, dout = self.rh_svc.readDigitalOutput()
   1557         retList = []
   1558         for item in dout:

/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/RobotHardwareService_idl.pyc in readDigitalOutput(self, *args)
    273
    274     def readDigitalOutput(self, *args):
--> 275         return _omnipy.invoke(self, "readDigitalOutput", _0_OpenHRP.RobotHardwareService._d_readDigitalOutput, args)
    276
    277     def getStatus2(self, *args):

BAD_OPERATION: CORBA.BAD_OPERATION(omniORB.BAD_OPERATION_UnRecognisedOperationName, CORBA.COMPLETED_NO)
```

130s, 2017-06-27:
I have no idea about this error. A few thoughts:

- [readDigitalOutput](http://fkanehiro.github.io/hrpsys-base/df/d98/classpython_1_1hrpsys__config_1_1HrpsysConfigurator.html#a7322eed06c2bd19890498b635098fd97) and [writeDigitalOutput](http://fkanehiro.github.io/hrpsys-base/df/d98/classpython_1_1hrpsys__config_1_1HrpsysConfigurator.html#ab0beb81c5b130145524e97f5f5bd5fa1) are methods from upstream library hrpsys as I linked to the document of each. I would suggest we'll keep tracking this down on their repository. I [initiated a conversation for you](https://github.com/fkanehiro/hrpsys-base/issues/1170), so please go ahead continue there.
- Make sure other DIO-related methods work for you (e.g. [this tutorial](http://wiki.ros.org/rtmros_nextage/Tutorials/Using%20digital%20IO%20%28NXO%20only%29)). If they don't, you might have other hindering issues that need to be fixed.

I close this for now but if there'll turn out to be anything specific to this repo, please reopen.

closed 2017-06-27 by 130s

k-okada, 2017-06-28:
This problem seems very nextage specific problem for me.....

1. please provide all python output since you start ipython ....
2. when did you last see the robot moves gripper , and what did you do then.
3. please check with following command from your ubuntu client PC
```
rtls nextage:15005/
rtcat -l nextage:15005/RobotHardware0.rtc
```

antgarmun, 2017-06-28:
The full python output is provided attached in this message. It is the first time that I use the grippers and/or DIO on the robot.

ipython -i `rospack find nextage_ros_bridge`/script/nextage.py -- --host hiro --robot RobotHardware0 --port 15005 --modelfile /opt/jsk/etc/HIRONX/model/main.wrl
Python 2.7.3 (default, Oct 26 2016, 21:01:49) 
Type "copyright", "credits" or "license" for more information.

IPython 0.12.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with hiro:15005
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x4af5ab8> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x4af5ab8> isActive? = False 
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x4af5830>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x4af5758>
[hrpsys.py]    version  =  1.0
;;
;; Loading ImpedanceController < 315.1.9
;;
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x4af5ab8> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x4af5ab8> isActive? = False 
[hrpsys.py] simulation_mode : False
[hrpsys.py]   bodyinfo URL = file:///opt/jsk/etc/HIRONX/model/main.wrl
[hrpsys.py] creating components
[hrpsys.py]   eval : [self.seq, self.seq_svc, self.seq_version] = self.createComp("SequencePlayer","seq")
[hrpsys.py] create Comp -> SequencePlayer : <hrpsys.rtm.RTcomponent instance at 0x4afef38> (1.0)
[hrpsys.py] create CompSvc -> SequencePlayer Service : <OpenHRP._objref_SequencePlayerService instance at 0x4b0b1b8>
[hrpsys.py]   eval : [self.sh, self.sh_svc, self.sh_version] = self.createComp("StateHolder","sh")
[hrpsys.py] create Comp -> StateHolder : <hrpsys.rtm.RTcomponent instance at 0x4aedf80> (1.0)
[hrpsys.py] create CompSvc -> StateHolder Service : <OpenHRP._objref_StateHolderService instance at 0x4b0d7a0>
[hrpsys.py]   eval : [self.fk, self.fk_svc, self.fk_version] = self.createComp("ForwardKinematics","fk")
[hrpsys.py] create Comp -> ForwardKinematics : <hrpsys.rtm.RTcomponent instance at 0x4b0a5f0> (1.0)
[hrpsys.py] create CompSvc -> ForwardKinematics Service : <OpenHRP._objref_ForwardKinematicsService instance at 0x4af9488>
[hrpsys.py]   eval : [self.el, self.el_svc, self.el_version] = self.createComp("SoftErrorLimiter","el")
[hrpsys.py] create Comp -> SoftErrorLimiter : <hrpsys.rtm.RTcomponent instance at 0x4afea70> (1.0)
[hrpsys.py] create CompSvc -> SoftErrorLimiter Service : <OpenHRP._objref_SoftErrorLimiterService instance at 0x4b02758>
[hrpsys.py]   eval : [self.log, self.log_svc, self.log_version] = self.createComp("DataLogger","log")
[hrpsys.py] create Comp -> DataLogger : <hrpsys.rtm.RTcomponent instance at 0x4af5950> (1.0)
[hrpsys.py] create CompSvc -> DataLogger Service : <OpenHRP._objref_DataLoggerService instance at 0x4b0a098>
[hrpsys.py] connecting components
[rtm.py]    Connect sh.qOut - el.qRef
[rtm.py]    Connect el.q - RobotHardware0.qRef
[rtm.py]    Connect RobotHardware0.servoState - el.servoStateIn
[rtm.py]    Connect RobotHardware0.q - sh.currentQIn
[rtm.py]    Connect RobotHardware0.q - fk.q
[rtm.py]    Connect sh.qOut - fk.qRef
[rtm.py]    Connect seq.qRef - sh.qIn
[rtm.py]    Failed to connect None to [None]([None])
[rtm.py]    Connect seq.basePos - sh.basePosIn
[rtm.py]    Connect seq.baseRpy - sh.baseRpyIn
[rtm.py]    Connect seq.zmpRef - sh.zmpIn
[rtm.py]    Connect sh.basePosOut - seq.basePosInit
[rtm.py]    Connect sh.basePosOut - fk.basePosRef
[rtm.py]    Connect sh.baseRpyOut - seq.baseRpyInit
[rtm.py]    Connect sh.baseRpyOut - fk.baseRpyRef
[rtm.py]    Connect sh.qOut - seq.qInit
[rtm.py]    Connect RobotHardware0.q - el.qCurrent
[hrpsys.py] activating components
[hrpsys.py] setup logger
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = q to RobotHardware0_q
[rtm.py]    Connect RobotHardware0.q - log.RobotHardware0_q
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = tau to RobotHardware0_tau
[... 29 lines cut ...]
[hrpsys.py] , JntAnglesOfGr=rarm, INITPOSE[i]=[-0.6, 0, -100, 15.2, 9.4, 3.2], tm=7, wait=True
[hrpsys.py] , JntAnglesOfGr=larm, INITPOSE[i]=[0.6, 0, -100, -15.2, 9.4, -3.2], tm=7, wait=True
Out[3]: True

In [4]: b=robot.readDigitalOutput()
---------------------------------------------------------------------------
BAD_OPERATION                             Traceback (most recent call last)
/home/kwduser/ros/hydro/catkin_ws/devel/openCV_tests/<ipython-input-4-4ab7a4ba7bdd> in <module>()
----> 1 b=robot.readDigitalOutput()

/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.pyc in readDigitalOutput(self)
   1554         @return list of int: List of the values in digital input register. Range: 0 or 1.
   1555         '''
-> 1556         ret, dout = self.rh_svc.readDigitalOutput()
   1557         retList = []
   1558         for item in dout:

/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/RobotHardwareService_idl.pyc in readDigitalOutput(self, *args)
    273 
    274     def readDigitalOutput(self, *args):
--> 275         return _omnipy.invoke(self, "readDigitalOutput", _0_OpenHRP.RobotHardwareService._d_readDigitalOutput, args)
    276 
    277     def getStatus2(self, *args):

BAD_OPERATION: CORBA.BAD_OPERATION(omniORB.BAD_OPERATION_UnRecognisedOperationName, CORBA.COMPLETED_NO)

k-okada, 2017-06-28:
> /home/kwduser/ros/hydro/catkin_ws/devel/openCV_tests/ in ()

> configuration ORB with hiro:15005

Are you using hiro? Or nextage?
Anyway can you send us the picture of your robot and gripper ?

antgarmun, 2017-06-28:
I am using HIRO but I can't sent you pictures.

k-okada, 2017-06-28:
I'm afraid hiro does not have DIO systems

antgarmun, 2017-06-28:
I follow the robot manual and It has DIO systems. The manual talks about NEXTAGE robot however, in some places of the documentation that I have talks about HIRO. It is possible that the robot is a NEXTAGE but use some things from HIRO development. 

All works ok following the documentation but not the DIO system.

k-okada, 2017-06-28:
It seems you're trying to control root gripper via DIO https://github.com/fkanehiro/hrpsys-base/issues/1170#issuecomment-311574474, but 
 if you're really using HIRO robot (http://robotics.naist.jp/wiki/?plugin=ref&page=%E7%A0%94%E7%A9%B6%E8%A8%AD%E5%82%99%2FHIRO-NX&src=HIRO-NX.jpg), you need  to use https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L473-L513, because the gripper is controlled via servo motor (and serial port), not DIO.
If you're using Nextage (http://nxt.typepad.jp/.a/6a0120a6ffc261970b019b00b83582970c-pi), the gripper is connected via DIO.
Anyway you'd better to contact manufacturer or service provider, we can not judge the situation without picture of the robot.

130s, 2017-06-28:
> I follow the robot manual and It has DIO systems.

Just to be clear, is that a hardcopy or is it online? If online can you send the URL?

- I might be wrong but I'd be a bit surprised if the manufacturer's manual mix Hiro and NEXTAGE in a single document (still not impossible though).
- If you're referring to anything under http://wiki.ros.org/rtmros_nextage/Tutorials, unfortunately it's possible that available options for Hiro and NEXTAGE are often not clearly identified.

By default NEXTAGE has the letter on its chest it's NEXTAGE as in the picture in https://github.com/tork-a/rtmros_nextage/issues/343#issuecomment-311679243 unless the sticker is ripped off etc.

antgarmun, 2017-06-29:
The manual is a hardcopy which was provided us with the robot. There is a sticker in the robot's chest that says: "HIRO", but the manual has the name of NEXTAGE, the robot works with NEXTAGE commands and it's simulation model is NEXTAGE. On the other hand, the manual talks about HIRO in some parts and some commands are refered to HIRO too.

antgarmun, 2017-06-29:
The manual talks about DIO system and the API commands for control he hands are: 
In [8]: robot.h
robot.hand_width2angles  robot.handtool_l_eject   robot.hes_version
robot.handlight_both     robot.handtool_r_attach  robot.hgc
robot.handlight_l        robot.handtool_r_eject   robot.hrpsys_version
robot.handlight_r        robot.hes                
robot.handtool_l_attach  robot.hes_svc