tork-a/rtmros_nextage #345: Building rtmros_nextage in ROS Kinetic environment

JKBehrens, 2017-08-01:
Hello there!

Has anyone tried to build and use the rtmros_nextage with ROS Kinetic?

Thank you!

130s, 2017-08-03:
To my knowledge, a dependent package suite, [rtmros_common](http://wiki.ros.org/rtmros_common), isn't released into / confirmed to build on ROS Kinetic. In other words, if you can build `rtmros_common` on Kinetic then there should be no problem for `rtmros_nextage` itself to build on K.
Related https://github.com/start-jsk/rtmros_hironx/pull/506

130s, 2017-08-15:
Update: [Kinetic binary is now released](https://discourse.ros.org/t/new-packages-for-kinetic-2017-08-14/2447/1).

130s, 2017-08-15:
> Update: [Kinetic binary is now released](https://discourse.ros.org/t/new-packages-for-kinetic-2017-08-14/2447/1).

Looks like binary of `rtmros_nextage` isn't yet released. I assume binary of all dependency is available.

k-okada, 2017-08-18:
hi
Please check latest source tree including (https://github.com/tork-a/rtmros_nextage/pull/347), that should at least comparable on kinetic.