tork-a/rtmros_nextage #346: Catch exception error from nextage_client

choyos, 2017-08-09:
Hello there,
I was wondering if there are any option of handle exception error raise when there is a conflict in the trayectory of the Hiro movement. 
What I think if there is this chance, maybe I can create a function to restart robot position, not having to manually fix this issue.

Thanks for your attention.

k-okada, 2017-08-24:
hi, sorry for late reply, could you explain more detail on your problem? what is your input and output, etc.

choyos, 2017-08-24:
Hi, that's ok.
In this time I have forgotten was I was trying exactly, but I guess that my idea was to have a method to handle exception error when the Hiro robot have a movement issue.
Example: I have developed a ros node in order to control left arm movement, in order to follow objects, detected by another node that handle images (opencv detection). If the position of the arm is out of limits or "crash" with the robot torso (or the other arm or something), all servos go off for secure reasons, and the robot starts beeping, until I tell it (nxc.servoOff()).
If there is a chance to handle this "crash" by code, I will be able to automatically recover the robot. 
Sorry I cannot add any code or input output, but I think is more a generic question.