tork-a/rtmros_nextage #348: Problem with launching NEXTAGE Open (The real robot, not the simulation)

Robotawi, 2017-08-17:
I am using NEXTAGE Open robot. I have gone through all the beginner, and intermediate tutorials of [rtmros_nextage](http://wiki.ros.org/rtmros_nextage/Tutorials) I understood well about Moveit, and was able to use its IK_Fast and its commands also to control the simulated robot.

My Problem is that I still have a problem with running the real robot, not the simulated. 
I followed some closed issues and found that this command will be satisfactory:
 roslaunch nextage_ros_bridge nextage_ros_bridge_real_hrpsys315.2.7-.launch

However, it still doesn't work. It gives this output:
> raessa@nxconsole:~$ roslaunch nextage_ros_bridge nextage_ros_bridge_real_hrpsys315.2.7-.launch
... logging to /home/raessa/.ros/log/c4be22a6-8360-11e7-a101-000bab892524/roslaunch-nxconsole-9087.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nxconsole:53294/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtmnameserver_port: 15005
 * /rtmnameserver_robotname: RobotHardware0

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    rqt_hironxo (rqt_gui/rqt_gui)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
[... 575 lines cut ...]
[rtmlaunch]  This is very weird situation, Please check your network
[rtmlaunch] configuration with `ifconfig` on both robot and client side. 
[rtmlaunch]  Having multiple network interface sometimes causes problem, 
[rtmlaunch] please see FAQ site http://www.openrtm.org/OpenRTM-aist/html/FAQ2FE38388E383A9E38396E383ABE382B7E383A5E383BCE38386E382A3E383B3E382B0.html#f2bc375d
[rtmlaunch]            Issue related to this https://github.com/start-jsk/rtmros_hironx/issues/33
[rtmlaunch]            ~/.ros/log may contains usefully informations
[rtmlaunch] Connected from nextage:15005/RobotHardware0.rtc:q
[rtmlaunch]             to nextage:15005/CollisionDetector0.rtc:qRef
[rtmlaunch]           with {'id': '', 'properties': {'dataport.publisher.push_policy': 'all', 'dataport.subscription_type': 'new'}, 'name': None, 'verbose': False}
[rtmlaunch] collision_detector.launch did not work on both OpenRTM 1.0.0 and 1.1.0 
[rtmlaunch]    OpenRTM 1.0.0 Failed to make connection: Bad parameter 
[rtmlaunch]    OpenRTM 1.1.0 connect_ports() takes at most 3 arguments (6 given) 
[rtmlaunch]  This is very weird situation, Please check your network
[rtmlaunch] configuration with `ifconfig` on both robot and client side. 
[rtmlaunch]  Having multiple network interface sometimes causes problem, 
[rtmlaunch] please see FAQ site http://www.openrtm.org/OpenRTM-aist/html/FAQ2FE38388E383A9E38396E383ABE382B7E383A5E383BCE38386E382A3E383B3E382B0.html#f2bc375d
[rtmlaunch]            Issue related to this https://github.com/start-jsk/rtmros_hironx/issues/33
[rtmlaunch]            ~/.ros/log may contains usefully informations
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[INFO] [WallTime: 1502983807.808556] check 0 collision status, 0.000000 Hz
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] check connection/activation

k-okada, 2017-08-18:
```
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7f9f26a1e6c8>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x7f9f26a1e5f0>
[hrpsys.py]    version  =  315.2.8
[hrpsys.py]   bodyinfo URL = file:///opt
```
tells your robot has version 315.2.8, so try `nextage_ros_bridge_real.launch` ?

Robotawi, 2017-08-18:
Thank you for your reply.
With `nextage_ros_bridge_real.launch`
I get this: 
```
raessa@nxconsole:~$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch
... logging to /home/raessa/.ros/log/712a9e36-842c-11e7-a371-000bab892524/roslaunch-nxconsole-4695.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nxconsole:54624/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtmnameserver_port: 15005
 * /rtmnameserver_robotname: RobotHardware0

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    rqt_hironxo (rqt_gui/rqt_gui)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
[... 523 lines cut ...]
[rtmlaunch]           with {'id': '', 'properties': {'dataport.publisher.push_policy': 'all', 'dataport.subscription_type': 'new'}, 'name': None, 'verbose': False}
[rtmlaunch] hrpsys_ros_bridge.launch did not work on both OpenRTM 1.0.0 and 1.1.0 
[rtmlaunch]    OpenRTM 1.0.0 Failed to make connection: Bad parameter 
[rtmlaunch]    OpenRTM 1.1.0 connect_ports() takes at most 3 arguments (6 given) 
[rtmlaunch]  This is very weird situation, Please check your network
[rtmlaunch] configuration with `ifconfig` on both robot and client side. 
[rtmlaunch]  Having multiple network interface sometimes causes problem, 
[rtmlaunch] please see FAQ site http://www.openrtm.org/OpenRTM-aist/html/FAQ2FE38388E383A9E38396E383ABE382B7E383A5E383BCE38386E382A3E383B3E382B0.html#f2bc375d
[rtmlaunch]            Issue related to this https://github.com/start-jsk/rtmros_hironx/issues/33
[rtmlaunch]            ~/.ros/log may contains usefully informations
[rtmlaunch] Connected from nextage:15005/RobotHardware0.rtc:servoState
[rtmlaunch]             to nextage:15005/HrpsysSeqStateROSBridge0.rtc:servoState
[rtmlaunch]           with {'id': '', 'properties': {'dataport.publisher.push_policy': 'all', 'dataport.subscription_type': 'new'}, 'name': None, 'verbose': False}
[rtmlaunch] hrpsys_ros_bridge.launch did not work on both OpenRTM 1.0.0 and 1.1.0 
[rtmlaunch]    OpenRTM 1.0.0 Failed to make connection: Bad parameter 
[rtmlaunch]    OpenRTM 1.1.0 connect_ports() takes at most 3 arguments (6 given) 
[rtmlaunch]  This is very weird situation, Please check your network
[rtmlaunch] configuration with `ifconfig` on both robot and client side. 
[rtmlaunch]  Having multiple network interface sometimes causes problem, 
[rtmlaunch] please see FAQ site http://www.openrtm.org/OpenRTM-aist/html/FAQ2FE38388E383A9E38396E383ABE382B7E383A5E383BCE38386E382A3E383B3E382B0.html#f2bc375d
[rtmlaunch]            Issue related to this https://github.com/start-jsk/rtmros_hironx/issues/33
[rtmlaunch]            ~/.ros/log may contains usefully informations
[ WARN] [1503071233.319023167]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is not executed last 5[sec]

```

k-okada, 2017-08-22:
please check your network connection, if you have two ether port, connect robot to eth0
```
[rtmlaunch]  This is very weird situation, Please check your network
[rtmlaunch] configuration with `ifconfig` on both robot and client side. 
[rtmlaunch]  Having multiple network interface sometimes causes problem, 
```

Robotawi, 2017-08-22:
Thank you for your reply.
The problem is still, and here are the details.
I traced [this port](http://i.imgur.com/l6BJMsN.jpg) which connects NEXTAGE OPEN to the LAN and found that there are two LAN ports in the interface computer as shown [here](http://i.imgur.com/J2BkchE.jpg). The shown is the default connection, in this case, I can ping nextage host and it is responding but the previously mentioned relaunch commands are not working.
```
roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch
roslaunch nextage_ros_bridge nextage_ros_bridge_real_hrpsys315.2.7-.launch
```
When I changed the connector [position](http://i.imgur.com/3ZOoXmk.jpg), I totally lost the LAN connection with NEXTAGE as shown [here](http://i.imgur.com/tHBVVuj.jpg). Now, even ping doesn't work. Any suggestions?

Thank you

k-okada, 2017-08-22:
use `ifconfig` on ubuntu machine to see current configuration

longjie, 2017-08-23:
In my memory, NXO has a small ethernet hub in the component. Can you see it ?
And with the factory default, the controllerbox(QNX) and PC(ROS) is on the same private network, 192.168.xxx.xxx(I don't remember). I think the ehternet port on the backside is also connected to this private (static) network.

I remember we have to be careful to change the network, when we need to log on to the ROS PC from your remote machine. With the factory setting, it should work just fine. I don't know how your network is configured now. So, as the next step, you should check your configuration with ifconfig as @k-okada memtionned.