tork-a/rtmros_nextage #370: [nextage_gazebo] World is very dark on melodic

pazeshun, 2020-04-17:
On melodic:
```
$ roslaunch nextage_gazebo nextage_world.launch
```
[image]

On kinetic:
```
$ roslaunch nextage_gazebo nextage_world.launch
```
[image]

pazeshun, 2020-04-17:
When I added many lights:
[image]

k-okada, 2020-04-17:
yeah, the world is dark, but the object looks good, and the PointClouds seems ok

I first thought this is because I prefer super heavy "mobile workstation" (with GPU) and @pazeshun likes thin and light laptop (without GPU?)... Partially it is true. The robot itself seems dark on my GPU machine and bright on T480s. But the color of objects in Gazebo and PointClouds seems ok.

T480s (see glx log https://gist.github.com/k-okada/ff01c0a2755e5d35e21612c571c283a4 )
[image]

k-okada, 2020-04-17:
And it is always better to check collade file, that is very hard to write a program that correctly interprets the model file, because the collad file is so flexible that there are many way of describe the model. So if we convert the .dae to .stl , we will get something like this;

[image]

https://gist.github.com/k-okada/057f391d7fcab4a0ebb1bd3a182f839f

pazeshun, 2020-04-17:
I confirmed [your tutorial](https://github.com/start-jsk/rtmros_tutorials/pull/572) works and color of object can be acuquired from point cloud (I also tried blue box and red box, there seemed no problem).
However, I think light issue should be fixed, because Nextage itself seems black even in RGB image.
[image]

k-okada, 2020-04-17:
I see ,from https://bitbucket.org/osrf/gazebo/issues/2623/no-shadows-and-sun-light-with-non-nvidia, what we can do is to disable shadows ...

see
https://github.com/k-okada/rtmros_tutorials/commits/disable_shadow
for 'sample.world' settings for disabling shadows.

[image]

pazeshun, 2020-04-17:
Wow, https://github.com/k-okada/rtmros_tutorials/commit/659d630a27d34230ea3aa906d0297e3d57140bec makes the dawn!!!
[image]
My PC is T460s including NVIDIA GPU, but I don't activate it.

the-raspberry-pi-guy, 2020-04-20:
Hi! Working on something similar here - how did you get the Nextage Open model in melodic? All of the docs point towards older versions of ROS and I am struggling here! Any advice appreciated (relatively new user here) :)

pazeshun, 2020-04-22:
@the-raspberry-pi-guy 
Sorry for late, and I feel you can work Nextage now according to #373 .
But I'm writing information just in case.

Now our rtmros base software is released on [ROS testing repository](http://wiki.ros.org/TestingRepository).
If you already use that repository, what you should have to build from source is [rtmros_hironx](https://github.com/start-jsk/rtmros_hironx) and rtmros_nextage.
All other dependencies should be installed via `rosdep install` to those repositories.
Currently, #365 is required to build rtmros_nextage on melodic.
If you don't want to use ROS testing repository and cannot wait about two weeks (delay to find mistakes), you have to build [hrpsys](https://github.com/fkanehiro/hrpsys-base), [rtmros_common](https://github.com/start-jsk/rtmros_common), rtmros_hironx, and rtmros_nextage at least.

the-raspberry-pi-guy, 2020-04-22:
Hi @pazeshun, thanks for your response :) I did indeed manage to get the Nextage Open model to load in Gazebo - thank you for your advice with this.

Do you have any advice for issue #373? The model seems to be not calibrated properly - I am getting a lot of vibration when trying out movements.