tork-a/rtmros_nextage #373: [nextage_gazebo] Tuning & Model Vibration Issues

the-raspberry-pi-guy, 2020-04-21:
I am currently trying to work with the Nextage platform in Gazebo, and facing issues with the model and Gazebo world. I wrote a quick program to test the range of each joint of the right arm on the Nextage. When running these on each joint, I get considerable vibration and jittery/unpredictable behaviour as if the model has not bee tuned/setup correctly. 

Please see this video for my findings: https://www.youtube.com/watch?v=xUfgR7gipZo&feature=youtu.be (note that some of the limits for joints close to the end effector are different to the ones I tested)

Does anyone know how to rectify these issues and get stable performance with the Nextage platform in Gazebo?

pazeshun, 2020-04-22:
@the-raspberry-pi-guy I'm a beginner of Gazebo and don't have detailed knowledge, but perhaps you should tune PID control gains.
They are defined at https://github.com/tork-a/rtmros_nextage/blob/indigo-devel/nextage_gazebo/config/nextage_gazebo_control.yaml

pazeshun, 2020-04-22:
If you find any good gains, please make a pull request of that.

the-raspberry-pi-guy, 2020-04-22:
I am also new to Gazebo, and have been trying to tune the PID gains. I'll let you know if I make any progress.

the-raspberry-pi-guy, 2020-04-26:
@pazeshun do you also have the same model vibration and tuning issues? What version of ROS and Gazebo are you using?

I am using the latest Melodic and Gazebo 9.0. I suspect that this unreliable behaviour is due to the fact that these docs were written for older versions of Gazebo.