tork-a/rtmros_nextage #375: [nextage_calibration] Problem of a dependency and modification

y-yosuke, 2020-07-13:
The debian package of rtmros_nextage for ROS Melodic was published, but I could not install this package on ROS Melodic by apt with a error of nextage_calibration dependency problem as shown below.

```
$ sudo apt install ros-melodic-rtmros-nextage
[sudo] password for robotuser: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-melodic-rtmros-nextage : Depends: ros-melodic-nextage-calibration but it is not installable
E: Unable to correct problems, you have held broken packages.
```

The problem is probably due to the lack of `turtlebot_description` in ROS Melodic debian packages.
And the package `turtlebot_description` was needed for Kinect depth sensor model.
Now there is a common package for Kinect sensor model in http://wiki.ros.org/openni_camera.
( https://discourse.ros.org/t/common-location-for-sensor-urdf-files/1758 )

So I changed the Kinect sensor model package to `openni_description` from `turtlebot_description` in the rtmros_nextage branch `mod-nextage-calibration-dependencies` https://github.com/tork-a/rtmros_nextage/tree/mod-nextage-calibration-dependencies .

But this branch still has a problem that  the `openni_description` debian package don't have the `mesh` folder yet as mentioned in PR https://github.com/ros-drivers/openni_camera/pull/72 . In the current situation where the PR is not merged, you need to clone the `openni_camera` package into your workspace to use the model.

If you have any better solutions, please tell me about them.

k-okada, 2020-07-14:
> ros-drivers/openni_camera#72

sorry, released new version https://github.com/ros/rosdistro/pull/25791
--
◉ Kei Okada

y-yosuke, 2020-07-15:
I also have checked the branch `mod-nextage-calibration-dependencies` in ROS Kinetic with cloning the `openni_camera` package and build it at my local workspace.

A screenshot of the result of executing `roslaunch nextage_calibration gazebo_kinect_checkerboard_waist.launch`

[image]