tork-a/rtmros_nextage #381: Python moveit_commander with Xarm6

Aki92-Xavier, 2021-10-05:
こんにちわ、現在moveitを用いてXarm6の動作計画をしようとしているのですが
チュートリアルを参考にXarm用に書き換えてみました。
まず、xarm_rosのパッケージより
roslaunch xarm6_moveit_config demo.launch　を起動後
書き換えたスクリプトを起動させましたが下記のようなエラーが出てしまいました。
[image]

参考チュートリアル:
https://rtmros-nextage.readthedocs.io/en/latest/manual_ja_tutorial_moveit-python.html#rtm-ros-bridge

書き換えたスクリプトは以下になります。
[test.txt](https://github.com/tork-a/rtmros_nextage/files/7284187/test.txt)

y-yosuke, 2021-10-05:
コンソールの warning や error，スクリプトを見る限り姿勢のクオータニオンがすべて 0 が代入されていることが原因のように思います．クオータニオンは長さ1の4次元の数ですので，すべて 0 ですと姿勢が定義できずにエラーになります．

- 参考: クオータニオンなど姿勢の指定方法
  - 手先の位置と姿勢を指定して動かす - TORK MoveIt Tutorial
    - https://github.com/tork-a/tork_moveit_tutorial/blob/indigo-devel/doc/moveit-tutorial_ja_robot-python_basic.md#%E6%89%8B%E5%85%88%E3%81%AE%E4%BD%8D%E7%BD%AE%E3%81%A8%E5%A7%BF%E5%8B%A2%E3%82%92%E6%8C%87%E5%AE%9A%E3%81%97%E3%81%A6%E5%8B%95%E3%81%8B%E3%81%99
    - https://github.com/tork-a/tork_moveit_tutorial/releases/tag/0.0.10

この原因の場合の解決策としては次の2つの方法が考えられます．

1. クオータニオンに正しい値を代入して姿勢を指定する
2. 姿勢の指定に ロール・ピッチ・ヨー の各角度を指定する

Aki92-Xavier, 2021-10-05:
ご返信ありがとうございます。 @y-yosuke さんがおっしゃる通りにクオータニオン、ロール・ピッチ・ヨーを適当な数値を指定することにより解を見つけることができたがrviz上では一瞬だけ残像のそうにアームが動作したことが確認できました。
その後下記のようなエラーを吐き、ターゲットポーズには行くことができませんでした。
[image]

y-yosuke, 2021-10-05:
見えている範囲のエラーはマニピュレータの「コントローラがない」と言っていて xarm は私は使ったことがないのでどれほどのことが分かるか分かりませんが，下記の点を確認願います．

- MoveIt の GUI でインタラクティブマーカーをマウス等で動かした場合の目標姿勢に動くことができるか？
- 制御対象は Gazebo シミュレータか実機か？
  - Gazebo シミュレータでは動作するが実機では動作しないということか？
- 画面に出ているエラーだけではなく，起動直後からの全てのコンソール出力を Issue に貼付
- Ubuntu と ROS のバージョン