tork-a/rtmros_nextage #389: How to connect point cloud to real robot?

Michi-Tsubaki, 2025-05-01:
Currently, frame_id of point cloud is `/camera`:
```
$ rostopic echo /points2 --noarr -n 1
header: 
  seq: 1028
  stamp: 
    secs: 1746096542
    nsecs: 243114999
  frame_id: "/camera"
height: 600
width: 800
fields: "<array type: sensor_msgs/PointField, length: 4>"
is_bigendian: False
point_step: 32
row_step: 25600
data: "<array type: uint8, length: 15360000>"
is_dense: False
---
```
However, `/camera` does not exist in TF tree of real robot, so we cannot see point cloud with robot model on RViz.
`stereo_link` is added in https://github.com/tork-a/rtmros_nextage/pull/109, but it is not connected to `/camera`.
How was `stereo_link` used previously?

cc. @pazeshun

Michi-Tsubaki, 2025-05-01:
In addition, CAMERA_HEAD_L and CAMERA_HEAD_R are statically defined in robot model for simulation. (https://github.com/tork-a/rtmros_nextage/issues/278)
Since for real camera use, usually in ROS, frame_id of left camera is used for stereo frame id, we want to use  CAMERA_HEAD_L as stereo_link. How should we do?

k-okada, 2025-05-09:
you can use https://wiki.ros.org/tf#static_transform_publisher, for example try
`rosrun tf static_transform_publisher 0 0 0 0 0 0 /CAMERA_HEAD_L /stereo_link 1000`
and if this works, run this node from your launch file.